OrcaSlicer/OrcaSlicer #1163: Nozzle Temperature Control

TechDude23, 2023-05-28:
We need a way to change the nozzle temperature with speed changes or layer height.

[Matte to Glossy Shift](https://forum.bambulab.com/t/matte-glossy-pla-shift/14428?u=3dtech )

Custom G-code works but we should have easy orca slicer solution.

Bambu Lab & others high speed printers.

All OS
1.6.3 Beta 

Temperature layer control 
[image]

julie777, 2023-05-29:
Does the height modifier allow changing temp?

TechDude23, 2023-05-29:
Hi Julie777, Yes it can be done by layer # or in mm using custom G-code in Orca if the Orca team can do something like the sample above by layer number or in mm height then you would have better control of the nozzle and bed temps. 

I will put some G-code samples up on Bambu community forum so we will have something for now to use :)

knw257, 2023-09-11:
I'd like this feature added because the built-in temperature tower in Orca has a base that's a bit too small, and keeps getting knocked over during printing, so I'd like an easy way to use alternative temperature towers and easily be able to set the nozzle temp per layer

straccio, 2023-10-10:
Hi to all. I'm interested in these kind of control.
I think that type of control must be made through volumetric change at temperature.

The need to increase the temperature of the nozzle is needed by high flow request to maintain "fluid" the material.

I immagine a test based on flow rate at different temperatures in order to have some fixed points of the curve and have a function that adapt the temperature of the nozzle on flow request.

closed 2024-01-17 by github-actions[bot]

Buzut, 2024-06-08:
I’d say this is even more needed to fight heat creep when flow is greatly reduced (bridging, overhang, ironing).

if we can’t get an automatic feature based on flow rate, considering adding it in the height range modifier would already by a huge life saver.

sb53systems, 2024-09-20:
Hi all,  
Here's a script created to prove the effectiveness of an automatic solution to this common problem..  
  
https://github.com/sb53systems/G-Code-Flow-Temperature-Controller  
  
I hope the Slicers will incorporate an equivalent solution!  
Happy smart 3D Printing.

carlosjln, 2024-12-02:
Right now I'm needing this feature to set a temp override within a height range modifier where only for this portion I've set faster printing speed for the inner perimeters and would benefit from improved flow.