OrcaSlicer/OrcaSlicer #1165: Increase solid infill area to next sparse infill line. AKA: Extend sparse infill

jeremytodd1, 2023-05-28:
PrusaSlicer 2.6 Alpha 5 introduced a great new feature: Extend sparse infill.
[Release notes for reference, scroll down to "Extend spare infill" section](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5)

Here is an image that nicely sums it up as well:
https://user-images.githubusercontent.com/10776877/32404178-27b5ffa0-c14b-11e7-98a9-3be600881306.jpg

Having this feature has honestly been a game changer for me recently, as you can now have much less infill but still have the internal solid bridging always connect to the infill.

It's the main thing preventing me from switching to OrcaSlicer. It'd be great to have this added.

SoftFever, 2023-05-29:
@jeremytodd1 
Thanks, this is good feature.
Any change you have model that can demonstrate this situation for me to test?

jeremytodd1, 2023-05-29:
> 

Honestly, this ends up applying to the vast majority of objects. 

As a quick example, I made a very small and simple test model:
https://i.imgur.com/0wwFt9P.png

Here is how OrcaSlicer handles the bridging infill for that upper part of the model:
https://i.imgur.com/JOdu1dV.png

You can see how a lot of the bridges aren't actually connected to anything so they will not print nicely at all.

Compare that to how this feature in PrusaSlicer will handle the same bridging infill:
https://i.imgur.com/lwFUXln.png

All of the ends of the bridges connect to the infill.

This is why I classify this feature as a gamechanger. It factors into practically any print job as it helps ensure the bridging infill prints as well as it can. 

You can also have insanely low infill percentage but still not need to worry about the internal bridging layers failing. Here is an extreme example of that same object but with very low infill:
OrcaSlicer: https://i.imgur.com/fGlkgQD.png
PrusaSlicer: https://i.imgur.com/1jlQ3J8.png

One of those objects will print without issue. The other object will have issues printing. _Thats_ what this feature does.

rfnovo, 2023-05-30:
@SoftFever there is an option in SuperSlicer called "Supporting Dense Layer" that solves this problem and on top of that still provides better top layer quality: 
[image]
[image]

It adds this sparse internal bridge before the actual internal bridge. It is quick to print and provide a good support for the internal bridge. Take a look here, were we had some discussion on that and I posted some pics of the difference in final print quality: https://github.com/SoftFever/OrcaSlicer/issues/1026

jeremytodd1, 2023-05-30:
@rfnovo @SoftFever 

Ah, thank you pointing that SuperSlicer feature out. It looks like it's a suitable solution. I haven't tried actually printing with it yet personally, but it looks like it will do essentially the same thing.

In my slicing testing for "Supporting dense layer", once thing I do want to point out is that I'm only having consistent reliable looking results with the "Automatic" algorithm option. The "Automatic, only for small areas", "Automatic, or anchored if too big", and "Anchored" options all give me unfavorable results at lower infills. 

The "Automatic" option is the only one that has been reliable no matter what the infill percentage is.

mlewis-everley, 2023-08-24:
@jeremytodd1 One item to note about "dense infill layer" from SuperSlicer is that it can (in many instances) be used with 0% infill. Essentially making it a quicker (and less consuming of filament) alternative to lightning infill.

Many features can be supported with just dense infill and nothing else, as an example I have attached something I print regularly:

[image]

[image]

[image]

You have to tune your bridging infill settings a bit, but once you have it is a great filament (and time) saver.

Noisyfox, 2023-09-16:
Orca (in fact, BBS) came with a settings called "Internal bridge support thickness" which does the similar job. Just set the value to your layer height or so and you will get extra supports under the bridges.

rfnovo, 2023-09-16:
I've it enabled and it does help indeed but not quite the same thing. Dense infill layer from SuperSlicer still more efficient IMO

Noisyfox, 2023-09-16:
[image]
Anyway I've done it.

blownupp, 2023-09-27:
@Noisyfox Excellent - looking forward to it! I just realized this feature was missing and came looking for any signs it existed/was in the works.

sb53systems, 2023-09-28:
Hello every one, (sory about my english)
I have a problem to control internal bridges thickness in orca slicer v 1.7.0, thin parameter dont make any changes !
[image]

Noisyfox, 2023-09-29:
> Hello every one, (sory about my english) I have a problem to control internal bridges thickness in orca slicer v 1.7.0, this parameter dont make any changes ! [image]
> 
> I hope some one can help me

You cannot change the thickness of ANY bridges because they are extrude over the air. The option in your screenshot doesn't change the thickness of the bridge, but change the thickness of EXTRA layers added below your bridge.

sb53systems, 2023-09-29:
> > Hello every one, (sory about my english) I have a problem to control internal bridges thickness in orca slicer v 1.7.0, this parameter dont make any changes ! [image]
> > I hope some one can help me
> 
> You cannot change the thickness of ANY bridges because they are extruded over the air and will ALWAYS be the same size as your nozzle size. The option in your screenshot doesn't change the thickness of the bridge, but change the thickness of EXTRA layers added below your bridge.

Thankyou @Noisyfox for your quick answer, in my case this makes the briding speed very slow and have a bad internal bridging, it's a limitation in this slicer ! i think that value must be the same as external bridges thikness to ensure a good bridging. and i hope this future wil be integrated in orcaslicer.
I would like to use orcaslicer but i'll stay using superslicer how gives me more control and beter prints.

Noisyfox, 2023-09-30:
> Thankyou @Noisyfox for your quick answer, in my case this makes the briding speed very slow and have a bad internal bridging, it's a limitation in this slicer ! i think that value must be the same [...]

Internal bridging will have a huhe improvement in the next release.

Noisyfox, 2023-11-07:
> > Anyway I've done it. A PR will be ready soon.
> 
> @Noisyfox so I am just trying this out (I have updated to 1.0b2), but I cannot find the settings you mentioned. My settings panel attached:
> 
> [image]

It doesn't have an option.

mlewis-everley, 2023-11-08:
> It doesn't have an option.

@Noisyfox So I have been testing this a bit and the dense infill appears to be being added under items that are printed in thin air, but there are no settings to control it. Is it just automatic in 1.8?

Noisyfox, 2023-11-08:
> > It doesn't have an option.
> 
> @Noisyfox So I have been testing this a bit and the dense infill appears to be being added under items that are printed in thin air, but there are no settings to control it. Is it just automatic in 1.8?

Yes it's automatically added when needed.

mlewis-everley, 2023-11-08:
> > > It doesn't have an option.
> > 
> > 
> > @Noisyfox So I have been testing this a bit and the dense infill appears to be being added under items that are printed in thin air, but there are no settings to control it. Is it just automatic in 1.8?
> 
> Yes it's automatically added when needed.

@Noisyfox That's cool! Thanks for the assistance.

igiannakas, 2023-12-27:
@SoftFever this can be marked as complete in the backlog, right?

tokamac, 2024-09-26:
> > > It doesn't have an option.
> > 
> > 
> > @Noisyfox So I have been testing this a bit and the dense infill appears to be being added under items that are printed in thin air, but there are no settings to control it. Is it just automatic in 1.8?
> 
> Yes it's automatically added when needed.

This is TERRIBLE for me. Enable it by default if you want, but please consider adding an _advanced setting to disable "Extend sparse infill"_.

See https://github.com/bambulab/BambuStudio/issues/4904
The issue is similar in OrcaSlicer. Not _exactly_ the same. For whatever reason the same 3MF will generate two unnecessary isolated breakable bridges and only one in OrcaSlicer. The issue remains though.

ianalexis, 2026-10-07:
https://www.orcaslicer.com/wiki/print_settings/strength/strength_settings_top_bottom_shells.html#surface-expansion

closed 2026-10-07 by ianalexis