OrcaSlicer/OrcaSlicer #1171: Device tab can't resolve hostname

kdb424, 2023-05-29:
**Describe the bug**
I configure my entire network to use machine hostnames so I can access http:/vzbot.home or http://v0.home and these show as able to connect in physical printer tab showing "Success". When going to the Device tab, it will not load and stays white. Switching to the machine's IP address in "Device UI" as opposed to the hostname causes everything to work properly.

**To Reproduce**
Set up new physical printer, use machine name to resolve it. Blank white Device tab.

**Expected behavior**
DNS addresses should be able to resolve to an IP address instead of requiring an IP that may be dynamic.

**Desktop (please complete the following information):**
 - OS: [MacOS/Artix Linux]
 - Version [Latest]
 - Orca Slicer Version [1.6.2]

tahaairplane, 2023-05-29:
same after i did the update its not working with the 1.6.3 nor 16.2

zerog2k, 2023-06-14:
also reproduced 1.6.3, macos 11.7.7
`http://10.0.4.39/` works, while `http://octopi/` does not

from terminal (10.0.0.1 is my router and also local dns resolver):
```
hackpro:~ jens$ host octopi
octopi has address 10.0.4.39
hackpro:~ jens$ dig +short @10.0.0.1 octopi
10.0.4.39
```

HumbleDeer, 2023-11-06:
I suspect this is related to the same kind of issue that's present in normal Chrome and Chromium. Non-TLD domains require a trailing forward slash (`/`) to work correctly. Perhaps this can help in the slicer too?

closed 2024-02-19 by github-actions[bot]