OrcaSlicer/OrcaSlicer #1172: Lost Custom Ender 3 Print and Filament Profiles but Printer Profile Remains

MiscPrinter, 2023-05-29:
**Describe the bug**
Since the update, 1.6.3, my Ender 3 profiles are lost. I have backed up the json file to my local drive prior to the update, tried to import but the import fails. Only the filament and print profiles are lost/won't import. The printer profile loaded back into the slicer after I restarted the software along with my X1C profiles. 

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Have custom Ender 3 filament and print profiles stored locally, File->Import->select the file->click open. The confirmation window shows 0 configs were imported (see screenshot). Additionally, the files were not retrieved from cloud backup as the printer profile was when the software was restarted post install. 

**Expected behavior**
The files should be imported and/or pulled from the cloud.

**Screenshots**
[image]
[image]

**Printer model**
Ender 3 Pro

**Desktop (please complete the following information):**
 - OS: MacOS 12.6.5
 - Version: Orca 1.6.3

jthe111, 2023-05-29:
This bug is not only on MacOS. This is happening on Windows 11 as well. Was occurring during the beta as well.

Kaan88, 2023-05-29:
Probably caused by this change: https://github.com/SoftFever/OrcaSlicer/pull/974

I had to modify json files to be able to import them.

tahaairplane, 2023-05-29:
same but mine is worse not only the profiles is GONE i cant connect to my printer as i did before

jthe111, 2023-05-29:
Which json files did you modify? I'm trying to locate them.

Kaan88, 2023-05-29:
I think it was the "inherits" section but I forgot. I think I installed the previous version, exported the configs then compared with 1.6.3 configs and looked for the differences. Since the configs were no longer there I had to use the projects I saved from before.

MiscPrinter, 2023-05-29:
I'm still able to access my Octoprint for the printer. It's just my profiles are gone and refusing to import. 

I don't see a "inherits" or a similar section. I see a "compatible_printers" section which has "Creality Ender-3 V2 0.4 nozzle", "Creality CR-10 Max 0.4 nozzle" in it. I remember having to add similar phrasing for my X1C profiles to allow them to be used for the P1P. See the profile I have tried and failed to import uploaded (remove the .txt extension. Github would not allow json file upload). 
[Standard E3P.json.txt](https://github.com/SoftFever/OrcaSlicer/files/11593503/Standard.E3P.json.txt)

MiscPrinter, 2023-05-29:
See #1182 for the hot fix for this issue.

closed 2023-05-29 by MiscPrinter