OrcaSlicer/OrcaSlicer #1173: icon missing in Windows system

botio, 2023-05-29:
**Describe the bug**
Keeping showing error windows about the icon missing, even though I reinstalled for many times with deleting /Appdata/roaming/orcaslicer

**3mf File for This Bug**
not related

**To Reproduce**
just reopen the app

**Expected behavior**
tell me how to reinstall correctly. Thank you

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10

SoftFever, 2023-05-29:
@botio 
Could you please try extracting OrcaSlicer to a regular directory (not in OneDrive) and try it again?

botio, 2023-05-29:
I did extract the folder on Desktop or Downloads (which is not in OneDrive) for sure. 

Just don't understand why even I redo the extraction at a different location still jumps out the error windows, even though I did delete everything in /Appdata/roaming/orcaslicer

hliebscher, 2023-05-29:
@botio 
Desktop and Downloads are shared in OneDrive

botio, 2023-05-29:
I did say I have turned the OneDrive down, and it still shows the error window right? In the "Describe the bug" part?

botio, 2023-05-29:
okay, find the way to fix it just by dragging the folder to D: .... wish there is a way to do the clean uninstalling.

closed 2023-05-29 by botio