OrcaSlicer/OrcaSlicer #1175: Texture for print bed is offset

Otann, 2023-05-29:
**Describe the bug**
When I load the texture for the print bed, the coordinate arrows are offset

**To Reproduce**
- Create the printer with a square bed (Voron 2.4 350x350 in my case)
- Load a square texture (1750x1750 px)

**Expected behavior**
- The coordinates start at the corner of the texture
- Like it does in PrusaSlicer
[image]

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: MacOS 12.6.1
 - Version 1.6.3

SoftFever, 2023-05-29:
@Otann 
Can you shared the texture? I can't repro with my texture(and the default Voron texture)

Otann, 2023-06-16:
of course

[image]

yourbuddydinec, 2023-07-01:
Same thing is happening to me with a custom image and seems the exact same offset in the y axis.

clovisd, 2023-09-23:
I'm able to see the custom Print Bed on my AMD Radeon GPU, however my scaling appears broken / offset.

Examples between Orca Slicer & Prusa Slicer with the same STL and bed SVG:
[image]
[image]

On the latest version of Orca Slicer vs latest version of Prusa Slicer.

clovisd, 2023-09-23:
This ticket also seems to show the offset issue with the XYZ green/blue/red arrows: https://github.com/SoftFever/OrcaSlicer/issues/1463

kluupom, 2023-11-02:
Same Here,
Bed is off, and in Addition no Textures work, so it is always as shown in [https://github.com/SoftFever/OrcaSlicer/issues/1463](#1643)
Just a Black Bed :(

Otann, 2024-02-23:
I believe this is still an issue. Should we just keep commenting here to provide an activity for github bot?

yourbuddydinec, 2024-02-23:
Yeah. Would be nice to get it fixed but I'm sure its low on the priority list. But good idea to keep letting em know it's still an issue

SoftFever, 2024-02-24:
Confirmed it's a bug

SoftFever, 2024-02-24:
Fixed in main branch.
[image]

The fix should be included in nightly build in about an hour from now: https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds

closed 2024-02-24 by SoftFever

Otann, 2024-03-21:
@SoftFever a hero! a legend!

commit d3309b97 (2024-02-24): Fixed a bug that caused bed textures to scale improperly
Fixes #1175