OrcaSlicer/OrcaSlicer #1177: Texture BED problem

iphonetik, 2023-05-29:
When I start Oscar Slicer, I set up my printer and then when the bed area where the parts are worked on appears, I see a black image as if it were a texture.

[image]
[image]

**Printer model**
Ender 3 V2 with Sprite Pro
 -Windows 10
 - Orca Slicer 1.63

iphonetik, 2023-05-29:
If I select a generic profile, it works fine. I cannot select the profile for my Ender 3 V2 because the issue occurs again.
[image]

fr3ak2402, 2023-05-29:
@iphonetik 

I have just done this and the texture appears normal to me.

[image]

But I also have to say that I already have the latest profiles, etc., because I'm working on them at the moment.

But actually the texture should already look like in your picture.

Can you please open the following folder in the OrcaSlicer installation folder and take a screenshot?

/resources/profiles/creality
In this folder there should be a file like the following:
creality_ender3v2_buildplate_model.stl
creality_ender3v2_buildplate_texture.png

iphonetik, 2023-05-29:
The files appear. I tried loading them manually, but nothing happens. I deleted the configuration folder and started over, but the same thing happens.

[image]
[image]

SoftFever, 2023-05-29:
@iphonetik 
What GPU do you have in your machine? Is it AMD?
I found some AMD driver didn't do well with non regular size texture.
Can you override the texture this modified textures and see if it works?
[creality_ender3v2_buildplate_texture.zip](https://github.com/SoftFever/OrcaSlicer/files/11592607/creality_ender3v2_buildplate_texture.zip)

iphonetik, 2023-05-29:
Yes i have RX480 GPU i try the new texture thanks!

iphonetik, 2023-05-29:
I solve the problem thanks!
[image]

iphonetik, 2023-05-29:
Thanks for support and Keep up the good work, I love Orca Slicer.

closed 2023-05-29 by iphonetik