OrcaSlicer/OrcaSlicer #1178: Network plugin problem

paajuf, 2023-05-29:
**Is your feature request related to a problem? Please describe.**
After unzip 1.6.3 orca-slicer and start app.. -> Problem with download plugin
**Describe the solution you'd like**
Can help with this problem ? 

WINDOWS 11 64
1.6.2 working without any problems .. 
[image]

mikedixonphoto, 2023-05-29:
> Having the same issues.... Found a workaround.

Well, care to share what the workaround was? I'm having the same problem, even after going back to 1.6.2

ccollinscj, 2023-05-29:
> > Having the same issues.... Found a workaround.
> 
> Well, care to share what the workaround was? I'm having the same problem, even after going back to 1.6.2

Yes,  I have the latest Bambu Slicer install 1.6.2.4  I simply copy the C:\Users\{your user name }\AppData\Roaming\BambuStudio\plugins to C:\Users\{your user name}\AppData\Roaming\OrcaSlicer\plugins which over wrote the bambu_network.dll and bambusource .dll which are newer dll's 4/26/2023.

closed 2023-11-20 by github-actions[bot]