OrcaSlicer/OrcaSlicer #1179: [Feature Request]: Submit OrcaSlicer to Winget Repository

kylek29, 2023-05-29:
**Describe the solution you'd like**
Winget is Microsoft's version a program repository and is growing in usage, it brings the simplicity of Linux's "apt/yum" command to download/install to the Windows OS. Along with that is a streamlined update process via the Winget update manager and user tools like WingetUI.

**Additional context**
Would simplify the update procedure for end users so they can maintain the latest version of the slicer. Integration is normally relatively simple (create manifest and push it to the Winget repository) and once published, can probably be automated in the Github CI/CD pipeline.

Package Manager Information:
https://learn.microsoft.com/en-us/windows/package-manager/package/

closed 2023-11-20 by github-actions[bot]

SheepCommander, 2024-05-09:
huh

robertbaker, 2025-03-21:
please reopen or keep my issue open