OrcaSlicer/OrcaSlicer #1180: Ringing tower test for calibration

The760, 2023-05-29:
Since you can't run a proper ringing test with a Bambu machine like you can on Klipper (where the acceleration increases by 500 as it gets higher), I think it would be great to include a customizable calibration that makes these adjustments to the g-code in the slicer. I want to know the max acceleration I can go on walls before ringing gets bad, and this would be a great solution to that.

Could just use Klipper's ringing tower test model, and prompt the user to input the start acceleration. It would be a really nice feature!

jeremytodd1, 2023-07-27:
Definitely agree.

For tuning new filaments, Orca currently has everything you need _except_ for calibrating for ringing. I think it'd be great if this was added.

DDFranks, 2023-07-27:
I agree, this would be a great help on tuning exterior accels after IS tuning

closed 2023-11-20 by github-actions[bot]

tlhintoq, 2023-11-20:
Wait - huh?
Went stale... But closed as 'completed'?
That seems weird to me.  As soon as I saw the issued I was in total agreement but it was closed out already.