OrcaSlicer/OrcaSlicer #1182: Update lost Filament and Process Profiles, can't create any new ones now.

eprouty, 2023-05-29:
**Describe the bug**
Extracted the new v1.6.3 files for Windows and over-wrote my existing install (probably my first mistake), started orca-slicer and found the profiles were missing, so I closed and restarted the application as mentioned in the instructions... still no luck.

Rebooted my machine, still no filament and process profiles, however my printer profiles are fine.

Tried to recover a profile my loading an old .3mf file and then saved the profile as a new "user preset" however this ALSO disappears upon restarting orca-slicer.

**To Reproduce**
1. Overwrite existing install with v1.6.3
2. Restart orca-slicer a few times with no luck recovering profiles.
3. Load .3mf project and try to save profile as user preset.
4. Restart orca-slicer and find no saved presets.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
Ender-3

**Desktop (please complete the following information):**
 - OS: Windows
 - Version v1.6.3

MiscPrinter, 2023-05-29:
See #1172 

It sounds like the same problem.

fr3ak2402, 2023-05-29:
@eprouty 
Your created profiles will be placed in the following folder,

C:\User\{username}\AppData\Roaming\OrcaSlicer\user

There should usually be a default folder there, in which all your settings that you have created are stored.

You have to adjust something in the JSON files so that everything works again. Due to the fact that the process name for some printers has been adjusted, this adjustment must also be made in your created processes etc. Then everything will be displayed normally again in OrcaSlicer.

eprouty, 2023-05-29:
Confirmed, this does indeed fix them.

For others that find this... open the folder mentioned above `C:\User\{username}\AppData\Roaming\OrcaSlicer\user` in an editor like vscode. It's the `inherit` section of the json files that is broken.

Do a find and replace for the following -

1. Find: `Generic PLA @Creality` -> Replace with `Creality Generic PLA` then do the same for PETG and ABS as needed
2. Find: `inherits.*@Creality` -> Replace with `@Creality Ender3V2` (or whichever profile you want to inherit from)

Hopefully this rough framework helps someone else out who finds this.  Would be nice if there wasn't a hard link between those strings causing the entire profile to be lost.

MiscPrinter, 2023-05-29:
The above find and replace in the json file is a hot fix for this issue.

jthe111, 2023-05-29:
> 1. Creality Generic PLA

Do I need to do this for each profile that I have made for multiple different fillaments? Or would doing this for the generics cause it to cascade down to all the filament profiles

eprouty, 2023-05-29:
> > 1. Creality Generic PLA
> 
> Do I need to do this for each profile that I have made for multiple different fillaments? Or would doing this for the generics cause it to cascade down to all the filament profiles

You need it for every custom profile you've created for both filaments and process. Each one will have an `inherit` key in the json that needs to match an existing generic profile.

Tyler8299, 2023-05-30:
Ok i got my filament presets back but the remainder of the code is missing for the user setting

Tyler8299, 2023-05-30:
[image]

Hunkoys, 2023-05-30:
Even my Printer profile was lost. What I did was I restored the previous install (It was in the recycle bin), then opened Orca Slicer twice to update the profiles. Renamed my old printer (was Ender 3 S1), then did the hotfix above.

If you permanently deleted your old OrcaSlicer, try window's "system restore".

fishercraigj, 2023-05-31:
Same issue on MacOS.  Commenting so I can get updates when this ticket changes.

iStarComLab, 2023-05-31:
Same issue on Macos 13.3.1(a) (22E772610a)
Presets from 1.6.2 did not pull up in 1.6.3
When saving ready-made presets, not one of them is saved from 1.6.3.
I rolled back to 1.6.2, all presets were pulled up automatically.
Make a fix to solve this bug in automatic mode, without manually editing all the presets?

Mr-Jay-420, 2023-06-01:
same issue with ender profiles on win10

JonLaRue, 2023-07-11:
I think it's a good idea to list **this as a known issue on the release page for V1.6.3**

zubamse, 2023-08-15:
I copiedIdea: https://github.com/SoftFever/OrcaSlicer/issues/1723#issuecomment-1669963834

madeintaiwan, 2023-08-19:
While the hotfix seems to work with filaments and possibly presets it looks like for some reason every custom printer (machine) profile is unusable. Luckily I have 3MF project files saved that have filament, machine and preset but this is a real disaster.

VulpinePrints, 2023-11-21:
This seems to be the best place to post my troubleshooting information from my instance of this bug in v1.8.0. I had all of these problems, with the exception that the "inherits" field in the JSON file would revert to blank. This was causing the filament profiles not to load. After multiple days looking for the answer on GitHub, as well as the greater internet, I was still at a loss. I am not a coder, but I have plenty of QC experience, so I started troubleshooting. I don't know WHY it caused the problem, but I think I understand WHAT caused the problem. 

Following info online trying to get syncing to work (big mistake), I reinstalled Orca to install the network plugin. Where the problem happened for me, was when I changed the install location to a new drive. Having performed a dozen reinstalls varying everything I could think of, I gave up and opened regedit and searched for Orca. For some reason, the installer did not update the location in the registry.

 I wiped everything again. This time I (backed up, then) deleted the registry information for both SoftFever and OrcaSlicer and restarted windows. I sought out any all references to Orca on my PC. When I was sure that there was nothing but my saved profiles. Then, I performed a fresh install (no network plugin). None of my saved profiles were present, but I could import all but the filament profile I had been working on most recently. I chose a generic filament and made my adjustments from the saved JSON file for that profile. I saved in a different location from my profile backups to save on confusion if I need it later. 

That was it. THAT was the cause. (Unconfirmed theory, remember not a coder). The screwy registry entries were causing OrcaSlicer to look in the wrong place when saving the calibrated profile. Since there was no profile for it to inherit from at that location, it set "inherits" as "" and no matter what changes were made to the file through Orca or via modifying the JSON file, the back end is looking in the wrong place because of bad registry information. Removing all references to Orca from the registry is not elegant, but at least for me, that worked. Hopefully someone more well versed in registry-fu can root out the offensive reference. 

There is a tell while editing the filament profile in Orca. Before I removed the registry entries, when I would build and save the filament profile I was working on, the start gcode and end gcode were always blank. Every generic filament profile had blank start and end gcode. After removing the registry references, they were not. I don't have a cause for certain. Hopefully this is enough to get the bug squashed and anyone dealing with it an easy-ish solution. 

I would like to point out that the files for the calibrated profiles still existed in the %AppData%\OrcaSlicer\user\default\(filament, machine, or process) folder. I got the information for my lost profiles by opening the folder with vscode and manually filled in a new profile from a generic profile after I fixed the registry errors. The info is there, the program just can't find or recognize it. 

I am aware of the ridiculousness of this hack-around, but perhaps this will give you an idea of how to diagnose and fix your problem. I'm sure that I have committed a thousand GitHub sins here. I'm new to this. I'm gonna need you to crawl all the way out of my butt. Seriously, though, I just hope I can save some other N00b 96 hours of hell. In my admittedly limited experience and opinion, OrcaSlicer is the best slicer available. I almost went back to Cura over this. It is the kind of thing that will run the less stubborn away to inferior options (opinion). I hope we get a fix in the next update. No one should have to manually enter their profiles 30+ times. 

For our siblings on MacOS, you are on your own. I hope my insights will give you a place to start. I haven't touched a Mac in over a decade. Linux users, you were on your own from the start and I bet that is how you like it. I have dabbled but never drank the Kool-aid. Good luck!

TL;DR - If you are experiencing profiles not saving after updating OrcaSlicer, 
1.  Backup your %AppData%\OrcaSlicer\user folder. 
2.  Uninstall anything to do with OrcaSlicer
3.  Remove the folder in %AppData%\OrcaSlicer
4.  RESTART the system to clear memory/cache
5.  Open regedit.exe
6.  ALWAYS back up the registry before edits. (file>export>enter backup name)
7.  Press CTRL + F and search for Orca
8.  Delete entries for OrcaSlicer and SoftFever (they should be consecutive)
9.  RESTART again
10. Reinstall OrcaSlicer from GitHub
11. Last RESTART
12. Run OrcaSlicer and follow the wizard
13. Install the BambuLabs Network Plugin at your own risk
14. Check that you can edit the generic profiles and that the start and end gcode have generated
15. Import your old config files from your backup from step 1. If you can't import a profile, view the JSON file in an appropriate viewer and manually copy the settings over.
16. Sell 3d printed items to everyone that has ever lived on or will ever live on the earth
17. Profit
18. Wake up and get back to work. Those Benchys aren't going to print themselves.

buildjournal, 2024-01-10:
Can someone please fix this in the next update? I'm trying to import printer, filament and settings for my K1 Max using [Guilouz's setup](https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/OrcaSlicer) but it is such a nightmare. I'm not a coder so I don't know what to change as suggested above.

Coop94, 2024-02-26:
If I don't have a C:\User{username}\AppData\Roaming\OrcaSlicer\user folder, do i just need to create one. I've edited a profile but maybe it stored the file somewhere else.

olehmykhalchenko, 2024-04-06:
It worked correctly on Orca v1.9.1, but after an update to v2, it reproduced for me. `Inherit` section of the JSON files looks broken, I need to delete Its value every time to see my filament profiles.

The-Monkey-King, 2024-04-22:
> Confirmed, this does indeed fix them.
> 
> For others that find this... open the folder mentioned above `C:\User\{username}\AppData\Roaming\OrcaSlicer\user` in an editor like vscode. It's the `inherit` section of the json files that is broken.
> 
> Do a find and replace for the following -
> 
>     1. Find: `Generic PLA @Creality` -> Replace with `Creality Generic PLA` then do the same for PETG and ABS as needed
> 
>     2. Find: `"inherits": "0.20mm Standard @Creality"` -> Replace with `"inherits": "0.20mm Standard @Creality Ender3V2"` (or whichever profile you want to inherit from)
> 
> 
> Hopefully this rough framework helps someone else out who finds this. Would be nice if there wasn't a hard link between those strings causing the entire profile to be lost.

Where do you find the exact profile name? In OrcaSlicer, the printer name is "Creality K1 Max (0.4 nozzle)" but when I look for that name I don't see it. However, I do see a json file -- K1 Max 0.4.json

olsonjj, 2024-06-11:
This is the most annoying bug. I've been hit by it twice now on my Mac.  I really like OrcaSlicer but this just annoying at this point.

flashburn2012, 2024-08-18:
Is this ever going to be fixed?

DuckersMcQuack, 2024-10-13:
> Is this ever going to be fixed?

October now, i'm on 2.2.0 beta, it still can't see my profiles.

[image]
[image]

k333bs, 2025-01-29:
This is still an issue for me.

zaza433, 2025-03-07:
Just updated from 2.1 to 2.2. Still an issue

moon-dad, 2025-06-01:
Going on the 2nd birthday for this one. Still an issue. GREATLY IMPACTS USABILITY OF THE SLICER

closed 2025-09-08 by github-actions[bot]