OrcaSlicer/OrcaSlicer #1183: Z Hop AKA Z Lift, Only Working On Bambu Printer profiles

PEZ18, 2023-05-29:
**Describe the bug**
I have Z Hop enabled in the Printer Settings/Extruder, but it does not work on any of my non Bambu Lab printers. It works fine on Bambu Lab profiles, and it even shows the spiral z lift tool path, but you dont see those same spiral z lifts on other machine printer profiles. 

I have even tried fresh printer profiles for my machine, other machines profiles, and different gcode flavors... No matter what I try, it only works if I take a Bambu Machine profile and edit the values such as bed size and other machine limits that match my printer settings.

**3mf File for This Bug**
N/A

**To Reproduce**
Enable Z Hop in Printer settings, set z hop height to .4mm, select spiral (or any type) in Z Hop type, and then slice file.

**Expected behavior**
I expect for the the Gcode to contain the proper commands to perform Z Hops after wipe/retractions, and I expect to see tool path showing the Z hop and the spiral movements as well.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]

[image]

[image]

Please let me know if I am just doing something wrong or missing something here. I never had this issue with my other slicers, so I dont know why this is happening on OrcaSlicer.

**Printer model**
I have a Tronxy X5SA Running latest Marlin 2.1.2 firmware. and I am on Windows 10 Pro
**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

djeZo888, 2023-05-29:
Is that since new version or are old versions also affected?

PEZ18, 2023-05-29:
> Is that since new version or are old versions also affected?

1.6.3, but it is the same issue with previous version as well. I dont think it has ever worked on any version to be honest. 

Test screenshot of 
[image]

PEZ18, 2023-05-29:
Looks like I just figured out how to get it to work, but I dont think this is the correct solution.

I can enable the G17 spiral movement along with Z Hop by going to filament overrides and enabling it there.

[image]

djeZo888, 2023-05-29:
Thats interesting, because Z Hop Spiral has definitely worked for my Kobra Max - made huge improvement with stringing and I could actually see it doing the funky move. I configured it in Machine settings (not Filament). That was on version 1.6.2.

Maybe it depends on how you configure other retraction parameters? Try to copy them as Bambu printer. I had my retraction settings almost the same as Bambu's stock ones.

PatrickDenis, 2023-05-31:
Hi Guy's, just to mention i found this thread by having the same issue with my Vzbot .. was trying to understand if there was something else to activate the Z hop since it wasnt working . 

I will try the setting overwrite section too.

Thanks

PEZ18, 2023-05-31:
@PatrickDenis I saw your reply in my email but its not here for some reason. I had same issue with the spiral lift as well, and it only worked if I used settings override in filament section of the settings. As you can see in my screenshots, there were no spiral movements before I changed the settings in the filament section, so spiral lift, and z hop are both not working unless you override them in filament section.

closed 2023-11-20 by github-actions[bot]