OrcaSlicer/OrcaSlicer #1186: Need the Slicing Tolerance option from Curas experimental settings

tjc2005, 2023-05-29:
Part tolerance is not good, I have multiple objects that won't fit together, however if I use cura on any of my printers and use exclusive mode in cura experimental settings, things work much better. With the same print settings.

Explanation here https://support.makerbot.com/s/article/1667418054188

The only way to get it to work for me currently is to make a part smaller or bigger but then that presents it's own set of issues with tolerances.

Can we get these added to orcaslicer?

rhiffsta, 2023-09-26:
i second this. a really useful option that is missing in orca slicer. please add these options.

PowerWiesel, 2023-12-03:
I would love to see that feature in Orca.
That's one of the reasons why I fallback to Cura sometimes

dbuezas, 2023-12-03:
Isn't this practically solved by x-y hole and contour compensation?

tjc2005, 2023-12-03:
> Isn't this practically solved by x-y hole and contour compensation?

No it's not the same unfortunately.

tjc2005, 2023-12-03:
> I mean, it used to fit only the .3mm tolerance, with x-y hole compensation set to 0.15mm it is spot on. <img alt="image" width="811" src="https://private-user-images.githubusercontent.com/777196/287542993-53977084-9a7a-4c1d-99e1-21493a3d297d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE2MzM1MzAsIm5iZiI6MTcwMTYzMzIzMCwicGF0aCI6Ii83NzcxOTYvMjg3NTQyOTkzLTUzOTc3MDg0LTlhN2EtNGMxZC05OWUxLTIxNDkzYTNkMjk3ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIwM1QxOTUzNTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMWYzZDJjNDJiNDgyODlhOTIwMzViZTRjZWVlMTA1MjQ3MDdjY2NiMGRlYzcxNzZjZGI0ZTVkMWJjYTFmMjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.e-tJYnRi5KM8Jqubgw1mwTNqAJEPRR80aTJV77nPEno">
> 
> (although I agree that Cura's tolerance seems less of a h
[image]
ack)

dbuezas, 2023-12-03:
I mean xy-hole (and xy contour) compensation, not xy-size compensation

dbuezas, 2023-12-03:
I'd suggest you do this:
1. print an orca tolerance test
2. see where the printed pin fits as tight as you wish tolerances to be
3. Divide the number by 2
4. Set it as xy hole compensation
5. Print the tolerance test again and corroborate it now fits in the zero position but it is not too loose

PowerWiesel, 2023-12-03:
Did you try this with threads? Just print a simple screw with idk 5mm OD and a nut and test if it works smoother.
It's probably also an issue with the tolerance of my printer, but on on my last design I wasn't able to get it screwed in more then one turn.
After I resliced and reprinted just one of both parts with that Cura setting it worked with almost no resistance. The difference is unbelievable.
The thread was also pretty big - 80mm OD. That might have been a factor.

markdid, 2023-12-14:
Was just looking for this feature

devinSpitz, 2023-12-14:
That is true for me as well as I print this design: https://www.printables.com/de/model/510256

Does work with Cura and "Slicing Tolerance" to "Exclusive" but not with the Orca slicer default settings.

after doing what  @dbuezas said it did work but it seems to be a workaround and not the same behavior. As +- nicely explained here I think: https://youtu.be/X6nCKQToOUg more precise here: https://youtu.be/X6nCKQToOUg?si=2XWAhxUQf3ybh-2x&t=201

Ardchie63, 2024-01-21:
The exclucive setting is very important for technical prints like screws or gears(heeringbone) without any test prints. It is one more needed step for designe, print, good result without playing.

AJolly, 2024-03-04:
This is the feature I really want from cura.  Makes screws so much easier.

p-dmitrij, 2024-05-02:
Very useful feature, it would be great to implement it!

lxlolxl, 2024-05-17:
I agree that this feature is necessary and urgent, please slice a 3mm diameter hole parallel to the XY plane (perpendicular to the printing wall), you will find that there will be a section of path occupying the round hole at the top and bottom, but if you use cura's "exclude" slicing mode, you won't have this problem, and it's not something that can be solved by the XY expansion and hole expansion.

arosovic, 2024-07-11:
> I'd suggest you do this:
> 
> 1. print an orca tolerance test
> 2. see where the printed pin fits as tight as you wish tolerances to be
> 3. Divide the number by 2
> 4. Set it as xy hole compensation
> 5. Print the tolerance test again and corroborate it now fits in the zero position but it is not too loose
> 
> And then just leave it like that for any print. That's what I like about this setting, it will look for holes and then correct for them by exactly that amount regardless of their size. I'm not sure but the value may differ for different line heights.

No, don't use hole compensation at all! (Use "Convert holes to polyholes" instead.)
It shoud be renamed to arc compensation, and reprogrammed accordingly because it has sense only for round holes, but it applies on square holes, too. It's sense should be to correct pulling filament towards the center of the hole, which is not present at square holes, but it is present at open arcs.

FireflyJar, 2024-09-02:
I am also looking for this feature.

Gismo6303, 2024-10-12:
Hello, I've switch for cura to orca today, and I need this feature too please

BigRon1955, 2024-10-27:
This feature is critical for producing accurate threaded parts.  Not understanding why this is a feature in Cura, but Orca Slicer and Bambu Studio have both failed to recognize the significance of this much needed setting.  I really like Orca Slicer, and don't like to be forced to use Cura for threads!

dbuezas, 2024-10-27:
Isn't it equivalent to "x-y hole compensation" and "x-y contour compensation"? Or rather, those are even more flexible.
I use those instead and don't have issues with threaded parts.

BigRon1955, 2024-10-27:
No, it's not equivalent.  X-Y hole compensation is very similar to
scaling.  Slicing tolerance is determined by the intersecting angle of the
slice/surface, which means it will not be applied uniformly as x-y hole
compensation would be.  Refer to the detailed Cura explanation, and
comments from others, including YouTube videos illustrating the
difference.  As a CNC programmer, I understand the way these offsets are
applied to surfaces when hemstitching 3D surfaces with a ball nose endmill.

rhiffsta, 2025-01-26:
this feature would be very useful. why there is still no activity after 1.5y from devs when several people show interest and also point out why it is useful/wanted.. could we please finally get a dev assigned to it or at least a statement of implementation plans?

AJolly, 2025-01-26:
Agreed, it's the one feature cura has that I really miss

antoniopegorelli, 2025-03-18:
It's a very nice feature for printers that are not as precise as Bambulab's models.
Even better if you want to print a downloaded STL and don't want to mess with scaling or redesigning to make it work.

closed 2025-06-25 by github-actions[bot]

Ardchie63, 2026-03-29:
On Top I prefer a different solution. It is a good idea to have 2 Parameter normal, exclude, include and only XY direction vs all XYZ