OrcaSlicer/OrcaSlicer #1187: Black build plates after adding different printer pack

axi0m89, 2023-05-29:
**Describe the bug**
On the Mecury one Zero G discord one of the members created a "pack" for the Zero G conversion. It contains printer images as well as the printer .json - This appears to be related to AMD drivers only and Prusa Slicer had the issue in 2022 but looks like the resovled it. https://github.com/prusa3d/PrusaSlicer/issues/8678 - This is affecting all build plates not just the ones used for the Zero G pack

**To Reproduce**
AMD drivers and then copy the files from the Zero G pack and put them in the resources folder.

**Expected behavior**
Build plates to remain normal

**Screenshots**
[image]

**Printer model**
Ender 5+ Converting to a Mecury one Zero G

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 22H2
 - AMD Graphics Driver 31.0.14057.2005 for the 6800XT

landomando, 2023-06-10:
Looks like it is related to the dimensions of the used texture. A temporary fix is to set the dimensions to a power of 2, such as 512 x 512 or 1024 x 1024. See also the [related comments](https://github.com/prusa3d/PrusaSlicer/issues/8417) for the Prusa Slicer issue.

axi0m89, 2023-06-22:
> Looks like it is related to the dimensions of the used texture. A temporary fix is to set the dimensions to a power of 2, such as 512 x 512 or 1024 x 1024. See also the [related comments](https://github.com/prusa3d/PrusaSlicer/issues/8417) for the Prusa Slicer issue.

Deleting the texture from the resources folder does fix the issue

VHelst, 2023-09-10:
Hi!
I´ve found another solution that was the only way for my buildplate to look right again:
I have a Sovol SV05, and I couldn´t find the file to delete in the resource-folder as mentioned here.
Instead I deleted the file "sovol_sv05_buildplate_texture.png" located in "OrcaSlicer_Windows_portable\resources\profiles\Sovol" and was very pleased with myself when I managed to fixe the issue...
I guess I just didn´t find it the first time I checked the resource-folder since the file is located 2 steps further into the foldertree, and that got me thinking that there may be other 3d-print owners out there that makes the same mistake I did, so I thought I post this as a suggestion on where to look for the file to delete...
Naturally I deleted the one that was meant for my printermodel (Sovol), and others will obviously have to go in to the folder that is used for their printerbrand...
Best regards
VHelst

zsmbrvr, 2023-09-23:
I fixed this issue by replacing the PNG file with an SVG one AND pointing the "bed_texture" to it.

closed 2023-12-31 by github-actions[bot]