OrcaSlicer/OrcaSlicer #1192: When printer is on LAN, auto start camera play

julie777, 2023-05-30:
**Is your feature request related to a problem? Please describe.**

Now that starting a print job automatically switches to the device page, which is nice, I want even more. :grin:.

I still have to press play to start the camera feed.

**Describe the solution you'd like**

Please auto start the live feed from the camera when the printer is not using the cloud. There is no penalty when using the LAN so it would be more convenient if it just started.

hliebscher, 2023-05-31:
That would only make sense if it is optionally selectable in the configuration. There are certainly some who may not want that feature, and it is specific to one manufacturer. Would it not be better to make this request to Bambu to ensure compatibility remains?

julie777, 2023-06-01:
@hliebscher 
I am still reporting bugs to BBL, if I can reproduce them with Bambu Studio. 
I am now submitting all feature requests here because BBL is never going to implement them and all I use now is Orca Slicer.

Make it a preference to I can choose it. If a printer supported by Orca does not have a camera then I would not expect it to work, or to have a camera feed showing in the device view.

closed 2023-11-20 by github-actions[bot]