OrcaSlicer/OrcaSlicer #1194: Command line execution does not work

julie777, 2023-05-30:
**Describe the bug**

I cannot find any way to run Orca Slicer from the command line.

According to documentation in the code if I provide invalid parameters on the command line it should show usage. It does not. I have tried the standard --help option and again nothing.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Run `orca-slicer --help` from the command line. 
Nothing happens.
There is no output.
The app does not incorrectly start the GUI.
There is no error code returned.

**Expected behavior**

I expect to see the command line help.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1.6.3

hliebscher, 2023-05-31:
@julie777  it works....  
OrcaSlicer --help
[2023-05-31 10:20:42.340883] [0x00000001f925db40] [trace]   Initializing StaticPrintConfigs
OrcaSlicer-1.6.3:
Usage: orca-slicer [ OPTIONS ] [ file.3mf/file.stl ... ]

OPTIONS:
 --datadir           Load and store settings at the given directory. This is useful for maintaining
                     different profiles or including configurations from a network storage.
 --debug level       Sets debug logging level. 0:fatal, 1:error, 2:warning, 3:info, 4:debug, 5:trace
 --load-filaments "filament1.json;filament2.json;..."
                     Load filament settings from the specified file list
 --load-settings "setting1.json;setting2.json"
                     Load process/machine settings from the specified file
 --outputdir dir     Output directory for the exported files.
 --skip-objects "3;5;10;77"
                     Skip some objects in this print
 --arrange option    Arrange options: 0-disable, 1-enable, others-auto
 --convert-unit      Convert the units of model
 --orient            Orient the model
 --scale factor      Scale the model by a float factor
 --export-3mf filename.3mf
                     Export project as 3MF.
 --export-settings settings.json
                     Export settings to a file.
 --export-slicedata slicing_data_directory
                     Export slicing data to a folder.
 --help, -h          Show command help.
 --info              Output the model's information.
 --load-slicedata slicing_data_directory
                     Load cached slicing data from directory
 --mstpp time        max slicing time per plate in seconds.
 --mtcpp count       max triangle count per plate for slicing.
 --no-check option   Do not run any validity checks, such as gcode path conflicts check.
 --normative-check option
                     Check the normative items.
 --pipe pipename     Send progress to pipe.
 --slice option      Slice the plates: 0-all plates, i-plate i, others-invalid
 --uptodate          Update the configs values of 3mf to latest.

Print settings priorites:
	1) setting values from the command line (highest priority)
	2) setting values loaded with --load_settings and --load_filaments
	3) setting values loaded from 3mf(lowest priority)

mydevpeeps, 2023-05-31:
It doesn't work for me either. I am on v1.6.3. OP didn't put in an OS. You never know when something you don't think is relevant, is. I am on Windows. @hliebscher what OS is that?

hliebscher, 2023-05-31:
@ mydevpeeps have you ran this from admin console ? in witch directory ?  and witch command?

mydevpeeps, 2023-05-31:
I was in the extracted directory, as Admin, at a command-prompt. I typed "orca-slicer.exe --help". A prompt came back with nothing.

[image]

julie777, 2023-06-01:
@hliebscher 

Not for me. 
I do not have an executable named OrcaSlicer. What OS are you on?

[image]

hliebscher, 2023-06-02:
Mac M1 OS 1.6.3 Nightlife

hliebscher, 2023-06-02:
@mydevpeeps have you seen anything in your log files?

mydevpeeps, 2023-06-03:
where are they?

Neekonu, 2023-09-29:
> **Describe the bug**
>
> I cannot find any way to run Orca Slicer from the command line.
>
> According to documentation in the code if I provide invalid parameters on the command line it should show [...]

@julie777 The subject of running CLI orca interests him.
Is it possible to split a file with a configuration loaded on the command line?

Noisyfox, 2023-11-08:
Unfortunately Windows requires a separate console build in order for the stdout to work properly when running from cmd.

thetazzbot, 2023-11-15:
Prusa slicer has a seperate executable in the same directory for slicing from the command line, called prusa-slicer-console.exe that does not appear to be present in orca slicer (or bambu studio).

kylek29, 2023-11-16:
Just tested this ... on Windows 10, running a CMD prompt as both admin and normal user, I get what everyone else was saying, just a blank line, and the next entry point.

However, if I pipe the output to a .txt file, the file *will* contain the appropriate menu. My guess is it's suppressing the output to the terminal when it sends it to stdout.

[image]

DiegoHerrera1890, 2024-02-16:
Are there any updates? Is it possible to use CLI in Ubuntu? I built successfully but I cannot run `orca-slicer --help`, it shows `command not found`.

grepson, 2024-03-03:
I can run the orcaslicer in cli mode, but slicing anything results in segmentation fault also. Running on MacOS 13.2.1 (22D68)

command:
```./OrcaSlicer --debug 6 --outputdir ~/Documents --export-3mf ./ouput.3mf --slice 0 ~/Documents/test.3mf```

output:

```
[2024-03-03 01:08:38.280521] [0x00007ff85400d680] [trace]   Initializing StaticPrintConfigs
[2024-03-03 01:08:38.292638] [0x00007ff85400d680] [warning] cli mode, Current OrcaSlicer Version 01.08.02.56
[2024-03-03 01:08:38.292668] [0x00007ff85400d680] [info]    Will start to read model file now, file count :1

[2024-03-03 01:08:38.292692] [0x00007ff85400d680] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0
[2024-03-03 01:08:38.292703] [0x00007ff85400d680] [info]    plate_to_slice=0, normative_check=1, use_first_fila_as_default=0
[2024-03-03 01:08:38.292763] [0x00007ff85400d680] [info]    read model file:/Users/macbookpro/Documents/test.3mf

[2024-03-03 01:08:38.292827] [0x00007ff85400d680] [info]    model 0x7ff7be868700, id 3, backup_path empty, set to /var/folders/hb/ql0w0r7971q0f5n4mntsw2br0000gn/T/orcaslicer_model/Sun_Mar_03/01_08_38#34298#3
[2024-03-03 01:08:38.292877] [0x00007ff85400d680] [info]    create /3D/Objects in "/var/folders/hb/ql0w0r7971q0f5n4mntsw2br0000gn/T/orcaslicer_model/Sun_Mar_03/01_08_38#34298#3"
[2024-03-03 01:08:38.293201] [0x00007ff85400d680] [info]    create /Metadata in "/var/folders/hb/ql0w0r7971q0f5n4mntsw2br0000gn/T/orcaslicer_model/Sun_Mar_03/01_08_38#34298#3"
[2024-03-03 01:08:38.293316] [0x00007ff85400d680] [info]    create /lock.txt in "/var/folders/hb/ql0w0r7971q0f5n4mntsw2br0000gn/T/orcaslicer_model/Sun_Mar_03/01_08_38#34298#3"
[2024-03-03 01:08:38.293558] [0x00007ff85400d680] [info]    _load_model_from_file:1559import 3mf IMPORT_STAGE_OPEN, m_load_restore=0

[2024-03-03 01:08:38.293658] [0x00007ff85400d680] [info]    _load_model_from_file:1595import 3mf IMPORT_STAGE_READ_FILES

[2024-03-03 01:08:38.776414] [0x00007ff85400d680] [info]    load_3mf found metadata key = Application, value = BambuStudio-1.9.1
[2024-03-03 01:08:38.776441] [0x00007ff85400d680] [info]    load_3mf found metadata key = BambuStudio:3mfVersion, value = 1
[2024-03-03 01:08:38.776451] [0x00007ff85400d680] [trace]   design_info, load_3mf found Copyright = 
[2024-03-03 01:08:38.776457] [0x00007ff85400d680] [info]    load_3mf found metadata key = Copyright, value = 
[2024-03-03 01:08:38.776466] [0x00007ff85400d680] [info]    load_3mf found metadata key = CreationDate, value = 2024-03-02
[2024-03-03 01:08:38.776474] [0x00007ff85400d680] [trace]   design_info, load_3mf found description = 
[2024-03-03 01:08:38.776479] [0x00007ff85400d680] [info]    load_3mf found metadata key = Description, value = 
[2024-03-03 01:08:38.776486] [0x00007ff85400d680] [trace]   design_info, load_3mf found designer = 
[2024-03-03 01:08:38.776491] [0x00007ff85400d680] [info]    load_3mf found metadata key = Designer, value = 
[2024-03-03 01:08:38.776497] [0x00007ff85400d680] [trace]   design_info, load_3mf found designer_cover = 
[2024-03-03 01:08:38.776503] [0x00007ff85400d680] [info]    load_3mf found metadata key = DesignerCover, value = 
[2024-03-03 01:08:38.776510] [0x00007ff85400d680] [trace]   design_info, load_3mf found designer_user_id = 
[2024-03-03 01:08:38.776516] [0x00007ff85400d680] [info]    load_3mf found metadata key = DesignerUserId, value = 
[2024-03-03 01:08:38.776522] [0x00007ff85400d680] [trace]   design_info, load_3mf found license = 
[2024-03-03 01:08:38.776527] [0x00007ff85400d680] [info]    load_3mf found metadata key = License, value = 
[2024-03-03 01:08:38.776534] [0x00007ff85400d680] [info]    load_3mf found metadata key = ModificationDate, value = 2024-03-02
[2024-03-03 01:08:38.776541] [0x00007ff85400d680] [trace]   design_info, load_3mf found model name = 
[2024-03-03 01:08:38.776546] [0x00007ff85400d680] [info]    load_3mf found metadata key = Origin, value = 
[2024-03-03 01:08:38.776553] [0x00007ff85400d680] [trace]   design_info, load_3mf found model name = 
[2024-03-03 01:08:38.776558] [0x00007ff85400d680] [info]    load_3mf found metadata key = Title, value = 
[2024-03-03 01:08:38.776667] [0x00007ff85400d680] [info]    _load_model_from_file:1746extract 1th file [Content_Types].xml, total=12
[2024-03-03 01:08:38.776686] [0x00007ff85400d680] [info]    _load_model_from_file:1839, [Content_Types].xml skipped, already parsed or a directory or not supported

[2024-03-03 01:08:38.776704] [0x00007ff85400d680] [info]    _load_model_from_file:1746extract 2th file Metadata/plate_1.png, total=12
[2024-03-03 01:08:38.777335] [0x00007ff85400d680] [info]    _extract_file_from_archive, extract  "/var/folders/hb/ql0w0r7971q0f5n4mntsw2br0000gn/T/orcaslicer_model/Sun_Mar_03/01_08_38#34298#3/Metadata/plate_1.png" from 3mf Metadata/plate_1.png, ret 1

[2024-03-03 01:08:38.777363] [0x00007ff85400d680] [info]    _load_model_from_file:1746extract 3th file Metadata/plate_1_small.png, total=12
[2024-03-03 01:08:38.777726] [0x00007ff85400d680] [info]    _extract_file_from_archive, extract  "/var/folders/hb/ql0w0r7971q0f5n4mntsw2br0000gn/T/orcaslicer_model/Sun_Mar_03/01_08_38#34298#3/Metadata/plate_1_small.png" from 3mf Metadata/plate_1_small.png, ret 1

[2024-03-03 01:08:38.777752] [0x00007ff85400d680] [info]    _load_model_from_file:1746extract 4th file Metadata/top_1.png, total=12
[... cut ...]
[2024-03-03 01:08:39.291622] [0x00007ff85400d680] [warning] no filament colors found in projects
[2024-03-03 01:08:39.291630] [0x00007ff85400d680] [info]    total 1 models, 0 objects

[2024-03-03 01:08:39.292733] [0x00007ff85400d680] [debug]   set_index: plate_id update from 33 to 0
[2024-03-03 01:08:39.292750] [0x00007ff85400d680] [debug]   set_index: plate_id update from 275779584 to 1
[1]    34298 segmentation fault  ./OrcaSlicer --debug 6 --outputdir ~/Documents --export-3mf ./ouput.3mf  0 
```

asokawotulo, 2024-03-24:
I'm getting the same issue as well when running the CLI. I narrowed it down to this line https://github.com/SoftFever/OrcaSlicer/blob/main/src/OrcaSlicer.cpp#L2777 when it attempts to load in the `extruder_clearance_max_radius` config.

I also noticed that this config is in a list of obsolete configuration keys [here](https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/PrintConfig.cpp#L5785) and no default value is provided.

----

Command: `orca-slicer --debug 5 --help`.

Output:
```
[2024-03-24 08:51:08.409324] [0x00007f75a0cf8c00] [trace]   Initializing StaticPrintConfigs
[2024-03-24 08:51:08.503534] [0x00007f75a0cf8c00] [warning] cli mode, Current OrcaSlicer Version 01.08.04.51
[2024-03-24 08:51:08.503625] [0x00007f75a0cf8c00] [info]    Will start to read model file now, file count :0

[2024-03-24 08:51:08.503735] [0x00007f75a0cf8c00] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0
[2024-03-24 08:51:08.503829] [0x00007f75a0cf8c00] [info]    plate_to_slice=0, normative_check=1, use_first_fila_as_default=0
[2024-03-24 08:51:08.503917] [0x00007f75a0cf8c00] [info]    run:before load settings, file count=0

[2024-03-24 08:51:08.504146] [0x00007f75a0cf8c00] [info]    current printer , new printer , current process , new process 
[2024-03-24 08:51:08.504210] [0x00007f75a0cf8c00] [info]    current printer inherits , new printer inherits , current process inherits , new process inherits 
[2024-03-24 08:51:08.504253] [0x00007f75a0cf8c00] [info]    old 3mf, no compatible printers, set to compatible
[2024-03-24 08:51:08.504311] [0x00007f75a0cf8c00] [info]    old printer , inherited from , old process , inherited from  ,compatible 1
[2024-03-24 08:51:08.504431] [0x00007f75a0cf8c00] [warning] no filament colors found in projects
[2024-03-24 08:51:08.504477] [0x00007f75a0cf8c00] [info]    total 0 models, 0 objects

[2024-03-24 08:51:08.510871] [0x00007f75a0cf8c00] [debug]   set_index: plate_id update from -1731686896 to 0
[2024-03-24 08:51:08.511000] [0x00007f75a0cf8c00] [debug]   set_index: plate_id update from 0 to 1
Segmentation fault (core dumped)
```

PR #4702 (2024-04-07): Fix segmentation fault when running CLI
Fixes: #1194

Changes:
- Replaced `extruder_clearance_max_radius` with `extruder_clearance_radius` to resolve the CLI error
- Add `if (m_plater)` before accessing preset_bundle in `generate_logo_polygon` and before `generate_plate_name_texture`
- Add thumbnails to export only if thumbnail callback is defined
closed 2024-04-07 by SoftFever