OrcaSlicer/OrcaSlicer #1197: Dev discussions is empty

julie777, 2023-05-30:
**Describe the bug**

It was mentioned to me that I should be participating in dev discussions. The only discussion under dev is about Portugese. 

Should some of the other discussions be tagged with "dev"?

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

hliebscher, 2023-05-31:
Please delete the unnecessary text in the bug reports that are provided. This will contribute to clarity, especially when English is not the native language. Everyone is free to start a discussion on a topic, but please do so under "Discussions". The "dev" label should be applied by the owner if they consider it a development-related issue. Ideally, only errors, bugs, and brief feature requests should be posted here. For more complex matters, please start a discussion.

julie777, 2023-06-01:
I posted it here because it is not a discussion but a request for a change. If you feel it is not appropriate then feel free to close it.

I would like to know how to find the discussions that are dev related so that I can participate.

closed 2023-11-20 by github-actions[bot]