OrcaSlicer/OrcaSlicer #1199: Unable to slice the Max Flow Rate model with a 0.6mm nozzle profile

mfattoru, 2023-05-30:
**Describe the bug**
I'm trying to tune the max flow rate of my filament. I have no issues when using a 0.4mm nozzle profile, but I'm unable to slice the model when using a 0.6mm nozzle.
I'm getting an error saying `Line width is too small`, and no matter how much I increase the value, the error doesn't go away

**3mf File for This Bug**
Max Volumetric Flow model

**To Reproduce**
- Set the printer to use a 0.6mm nozzle profile
- generate a max volumetric flow calibration
- An error will appear making it unable to slice and then print the model

**Expected behavior**
I can run max volumetric flow tests on a 0.6 nozzle (have not tested if other nozzles different from 0.4 have the same issue

**Screenshots**
[image]

**Printer model**
BambuLab P1P

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version [1.6.2+]

julie777, 2023-05-30:
@mfattoru 
Which process profile do you have selected?
Just coincidentally I am in the process of running calibrations for my filament with a .6 nozzle on my P1P. I just sent the max flow rate to the printer. I did not have any problems slicing. The profile that I had selected is a custom one that is based on the only .6 process profile shipped for the P1P. I cannot tell what I had changed at the moment because the test changed many settings.

I just noticed that you are running 1.6.2+ and I have installed the release version of 1.6.3

hliebscher, 2023-05-31:
can you upload the 3mf as zip for debug ?

julie777, 2023-06-01:
@mfattoru 
I would assume that the model generated is based on the printer/nozzle setting you have selected. You might also need to select a process profile for that nozzle size.

mfattoru, 2023-06-01:
I've added the used profiles [here](https://we.tl/t-Sc1mfNaW7m)

I've noticed that if I use the default bambu 0.3mm process file, the error does not appear. if I use my 0.2mm profiles instead, the error shows up.

hamster4430, 2023-06-16:
@mfattoru I have the same issue. A workaround were to print it with a standard 0.4 mm profile and then transfer the data to your edited one.
The actual point of the test is to find out the maximum possible flow rate. So it kinda works. I only had issues with overextrusion and blobs on bottom layers.

Probably the better solution were making a simple 3d object in a CAD software and slicing it in vase mode. I've done it unintentionally while I was trying to print something I needed urgently as I started increasing speed during the prining proccess.

hamster4430, 2023-06-22:
SOLUTION!

Error:
Line width too small
Jump to [SpeedTestStructure.step] (top_surface_line_width)

So the issue is obviously with a "line width" parameter. If you click on the slicer's error message it brings you to Object's Settings - Quality.
As you try to change:
**Line width:
Top surface**
it outputs a new notification and now it seems as if you can slice it:
"Spiral mode only works when wall loops is 1, support is disabled, top shell layers is 0, sparse infill density is 0 and timelapse type is traditional."
... but if you do slice it, you get another error:
"Invalid spacing supplied to Flow::with_spacing()"

The thing is that you also have to change it in Global settings.
------------------------------------------------------------------
Change it to 100-120% of your nozzles diameter (0.6-0.72mm) in following fields:

Global settings - Quality - Line width:
Default -> 0.6
First layer -> 0.6
Top surface -> 0.6

Object settings - Quality - Line width:
Default -> 0.6
Top surface -> 0.6
------------------------------------------------------------------
P.S.: You could save some time/material (5m36s) by disabling:
Object - Strength - Top/bottom shells:
Bottom shell layers -> 0

Global - Quality - Line width:
First layer -> 1.05

P.P.S.: Don't forget to repair non-manifold edges.