OrcaSlicer/OrcaSlicer #1201: Error when slicing with Tree Suppports: Flow::spacing() produced negative spacing. Did you set some extrusion width too small?

dvomang, 2023-05-30:
**Describe the bug**
When I use tree supports with OrcaSlicer I am getting an error that says 
"Flow::spacing() produced negative spacing. Did you set some extrusion width too small?"

**3mf File for This Bug**
3mf attached

**To Reproduce**
Unsure. Try to get Max Volumetric Flow to control the speed. Won't slice with supports on.

**Expected behavior**
I expected the slicer to work with supports.

**Screenshots**
[image]

**Printer model**
Elegoo Neptune 3 Pro on Klipper

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 1.6.3

dvomang, 2023-06-01:
This is resolved when I change the default line width from 0 to 0.6
[image]

5teveO, 2023-06-05:
I have has the same issue - when all widths are 0. Change to 0.4 (or probably anything) and its fine

scottmudge, 2023-06-18:
This PR fixes it.

closed 2023-11-20 by github-actions[bot]

MxBrnr, 2024-10-15:
This is still an issue as of 2.2.0-beta2. In my case it appears to be connected to using Archne and having the inner wall line width set to anything less than 110% of nozzle diameter.

Sa5-hA, 2026-01-04:
At the moment I am using orca slicer 2.3.1 and klipper I haven't noticed any errors like that anymore

jpluimers, 2026-09-27:
Had this happen in OrcaSlicer 2.4.2 with imported file "Configurable_Filament_Swatch_VZA_notext_PLA.stl" from https://www.printables.com/model/27814-filament-swatch using the "0.20mm Standard @Creality Ender3V2" profile and the (default) "Arachne" Wall Generator.

Switching to the "Classic" Wall Generator made the error go away.