OrcaSlicer/OrcaSlicer #1205: BBL X1C LAN Only mode not found if in different Subnet

fschramek, 2023-05-31:
**Describe the bug**
I cannot add a LAN Only Printer to the Device List, if the Printer is in a different Subnet from the PC with Orcaslicer. If I move the PC into the same subnet, then i can add the Printer to the list and after switching to the other subnet and restarting OrcaSlicer, the Printer is not in the List anymore. As it is now, you can not us the Printer in LanOnly mode if it is on a different subnet.

**To Reproduce**
Have the Printer in a different Subnet than the PC with OrcaSlicer. And have the LanOnly Mode activated on the printer.

**Expected behavior**
OrcaSlicer should be able to add an use the printer on a different Subnet.

**Printer model**
BambuLabs X1 Carbon

**Desktop (please complete the following information):**
 - OS: Windows10
 - OrcaSlicer Version 1.6.3

mydevpeeps, 2023-05-31:
Does the LAN Only mode use some kind of broadcast protocol that requires it to be "found"? If so you would need some kind of forwarder between subnets for that to work. Watching this one to see what happens.

hliebscher, 2023-05-31:
Each subnet is its own broadcast domain. Does it work with the original software? Or does it not work there either? Is there a route configured to the other network?

fschramek, 2023-06-01:
No it doesn't work with the Bambu Slicer either. There should be a route through the Router from one Subnet (VLAN ID 0) to the other (VLAN ID 4)

closed 2023-11-20 by github-actions[bot]

lukasmalkmus, 2025-01-20:
Late to the party but would be nice to have. Just stumbled across this.

spaceg00se-r, 2025-01-25:
I second this

Shadoweee77, 2025-02-02:
Yeah it also disconnects for me unlike Bambu Studio that remebers the device is in a different subnet.

spaceg00se-r, 2025-02-02:
Could a admin please reopen this issue?

spaceg00se-r, 2025-02-03:
@Shadoweee77 Do you know when this will be merged into the stable app?

Shadoweee77, 2025-02-03:
Release 2.3 I think but that's my educated guess ¯\_(ツ)_/¯