OrcaSlicer/OrcaSlicer #1207: updating nvidia gpu driver crashes running process

mydevpeeps, 2023-05-31:
**Describe the bug**
While running OrcaSlicer 1.6.3 I ran the NVIDIA Graphics Driver Update. After the driver finished running OrcaSlicer was gone (crashed).

**3mf File for This Bug**
N/A

**To Reproduce**

1. Open OrcaSlicer
2. Update Graphics Driver (NVIDIA Specific?)

**Expected behavior**
Running window re-renders like all the other apps running.

**Screenshots**
N/A

**Printer model**
N/A

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version 22H2 (build 19045.2728)

** Windows Event Viewer**
```
The description for Event ID 1 from source NVIDIA OpenGL Driver cannot be found. Either the component that raises this event is not installed on your local computer or the installation is corrupted. You can install or repair the component on the local computer.

If the event originated on another computer, the display information had to be saved with the event.

The following information was included with the event: 

The GPU has been disconnected and this application may become unresponsive.

Error code: 10
 (pid=60504 tid=29148 orca-slicer.exe 64bit)

Visit http://www.nvidia.com/page/support.html for more information.

The message resource is present but the message was not found in the message table

```

hliebscher, 2023-05-31:
Honestly, how often do you change graphics card drivers? Does it work fine afterwards?

mydevpeeps, 2023-05-31:
If you don't care close the issue as stupid/rejected. I don't need attitude for reporting an issue.

lbibass, 2023-05-31:
well yeah of course it would stop working. It's actively using the GPU. And you just took the GPU away. If you were running a game, updated the driver, and the game crashed, would you be mad at the game?

mydevpeeps, 2023-06-01:
I have many apps open that use gpu and this is the only one that crashed. back log it or close it.. either way your tone is not nessecarry.

julie777, 2023-06-01:
I posted about the NVidia problem with laptops having multiple GPUs. If you are in that category you should search for it. I am sure that searching issues for GPU will find it.

mydevpeeps, 2023-06-04:
I have a single RTX3070. Also my Orca Slicer keeps crashing while running it almost daily now. I'm trying to find the logs location. 

Not much that means anything to me in Event Viewer over and over again:

Faulting application name: orca-slicer.exe, version: 1.0.0.0, time stamp: 0x6472cf10
Faulting module name: BambuSource.dll_unloaded, version: 0.0.0.0, time stamp: 0x642f6637
Exception code: 0xc0000005
Fault offset: 0x0000000000087f01
Faulting process id: 0x9fa4
Faulting application start time: 0x01d996e8d2150271
Faulting application path: C:\Users\localadmin\bin\orca_slicer\orca-slicer.exe
Faulting module path: BambuSource.dll
Report Id: 7e1ecc8a-8e74-4168-aada-ab4175150b2a
Faulting package full name: 
Faulting package-relative application ID: 

Faulting application name: orca-slicer.exe, version: 1.0.0.0, time stamp: 0x6472cf10
Faulting module name: BambuSource.dll_unloaded, version: 0.0.0.0, time stamp: 0x642f6637
Exception code: 0xc0000005
Fault offset: 0x0000000000087f01
Faulting process id: 0x8938
Faulting application start time: 0x01d995928e37af3f
Faulting application path: C:\Users\localadmin\bin\orca_slicer\orca-slicer.exe
Faulting module path: BambuSource.dll
Report Id: f0c51e2d-2d5c-49c8-9185-019fea5b4b96
Faulting package full name: 
Faulting package-relative application ID: 

Faulting application name: orca-slicer.exe, version: 1.0.0.0, time stamp: 0x6472cf10
Faulting module name: BambuSource.dll_unloaded, version: 0.0.0.0, time stamp: 0x642f6637
Exception code: 0xc0000005
Fault offset: 0x0000000000083276
Faulting process id: 0x40d4
Faulting application start time: 0x01d9955f1be201a6
Faulting application path: C:\Users\localadmin\bin\orca_slicer\orca-slicer.exe
Faulting module path: BambuSource.dll
Report Id: ac55ea67-dd57-4837-893a-b8085ad1f29e
Faulting package full name: 
Faulting package-relative application ID:

closed 2023-11-20 by github-actions[bot]