OrcaSlicer/OrcaSlicer #1208: changing debug level to info crashes app

mydevpeeps, 2023-05-31:
**Describe the bug**
changed debug level logging to info at command line and on relaunch the app crashed. won't open until setting is removed.

**3mf File for This Bug**
N/A

**To Reproduce**
set command line option for debug to be info

**Expected behavior**
app is launched with info logging (where are those anyway?)

**Screenshots**
N/A

**Printer model**
Bambu X1C

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version 22H2

**Event Viewer**
```
Faulting application name: orca-slicer.exe, version: 1.0.0.0, time stamp: 0x6472cf10
Faulting module name: OrcaSlicer.dll, version: 0.0.0.0, time stamp: 0x6472cf06
Exception code: 0xc0000005
Fault offset: 0x0000000001639f5c
Faulting process id: 0x9388
Faulting application start time: 0x01d993ab2b2999dc
Faulting application path: C:\Users\localadmin\bin\orca_slicer\orca-slicer.exe
Faulting module path: C:\Users\localadmin\bin\orca_slicer\OrcaSlicer.dll
Report Id: c5d09c60-6d9e-4892-98ce-6b1e8951029a
Faulting package full name: 
Faulting package-relative application ID: 
```

hliebscher, 2023-05-31:
for MAC is ok
/OrcaSlicer --debug 5
[2023-05-31 14:26:02.914171] [0x00000001f925db40] [trace]   Initializing StaticPrintConfigs
loading network module at /Users/hliebscher/Library/Application Support/OrcaSlicer/plugins/libbambu_networking.dylib
after dlopen, network_module is 0x94794ab0
UNSUPPORTED (log once): POSSIBLE ISSUE: unit 0 GLD_TEXTURE_INDEX_2D is unloadable and bound to sampler type (Float) - using zero texture because texture unloadable
1. lo0   : 127.0.0.1
2. en0   : 192.168.20.46
3. utun5 : 192.168.2.215

*   Trying 140.82.121.6:443...
*   Trying 54.219.5.19:443...
* Connected to api.github.com (140.82.121.6) port 443 (#0)
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*

mydevpeeps, 2023-06-01:
ah. I used the work "info" and not a number. does the param input validate data types?

closed 2023-11-20 by github-actions[bot]