OrcaSlicer/OrcaSlicer #1214: Hi, so i switched from cura to orcaslicer with my ender 3s1. but sice the 1.6.3 version released there is a layershift at always the same hight. when i use cura thzere isn't so there has to be a bug in the slicer

Nefnief-tech, 2023-05-31:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
[3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11617021/3mf.zip)

 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

DKS-18, 2023-06-07:
more than likely it is either the acceleration settings in orca slicer are too high for your printer or the z hop isn't enabled or is a value too small that is causing the print head to clip the print and have a layer shift

PhilBaz, 2023-06-07:
@Nefnief-tech 

Yes, this is most likely not a slicer bug. But rather you have either a part of the print that's curling and catching the nozzle, or you have instructed Orca to outrun what your motors are capable of and are getting lost steps.

You should close this issue.

closed 2023-06-07 by Nefnief-tech