OrcaSlicer/OrcaSlicer #1216: Flow Rate Ignored

Highlander-bdj, 2023-05-31:
Filament flow rates are ignored unless "classic mode" for overhang speed is selected.

[image]

[image]

RydellV, 2023-06-24:
Can confirm this as well. Tried making a TPU profile by just limiting the volumetric flow rate to 2 and after slicing there were  several overhangs at 14. Classic mode fixed it.

Marijn1250, 2023-08-16:
Same here. I tried switching to classis mode and that fixed the flow issue. 
I then switched back to Arachne mode and sliced again. The problem was still fixed. 
Maybe you can test the same. 

- Go to Classic Mode and slice.
- Check preview - select Flow 
- Go back to Arachne Mode and slice
- Check the preview again - Flow

PrincessLuna2253, 2023-08-23:
Confirming that I am also seeing this issue in version 1.6.4. The workaround of slicing in classic mode and then turning classic mode off and slicing again does not work for me.

hatkyinc2, 2023-11-18:
Sounds like the same issue as https://github.com/SoftFever/OrcaSlicer/issues/2033

I've tried classic mode now with 1.8 beta2, doesn't seem to work

closed 2024-02-25 by github-actions[bot]