OrcaSlicer/OrcaSlicer #1217: OverHang Speed Issue when using 'Classic' Wall Mode

ProjectKerr, 2023-05-31:
Using Bambu P1P Printer - Issue only apparent on latest release of OrcaSlicer 

Using 'Classic' Wall Generator I get a weird Slow Down leading up to Overhangs, Disappears If I change to 'Arachne' or turn off 'Slow down for overhangs' 
First pic is Orcaclicer
Second Pic is BambuSlicer (same settings)
[image]
[image]

cpasoft, 2023-06-07:
I confirm this... 
[image]

Also, if you set the classic mode on the overhang speed section, something more weird happens...
[image]

cpasoft, 2023-06-08:
The same with arachnee and classic mode on
[image]

closed 2023-11-20 by github-actions[bot]