OrcaSlicer/OrcaSlicer #1221: Program crashes when trying to slice with a raft

jthe111, 2023-05-31:
**Describe the bug**
As the title describes, when trying to slice any files (both simple geometries and complex geometries) with a raft, the program just crashes right to desktop. It will slice up to 50% completed then crash. Sometimes it will throw up a "Access Permission" error and sometimes it crashes right to desktop.

**3mf File for This Bug**
Happens on various files. I have sliced various versions of cinderwings crystal dragons and a MTG boulder deck case since updating. This issued was never present on 1.6.2

**To Reproduce**
Grab a file, add any number of raft layers and various raft spacing options. It will commence to slice but then it will crash.

**Expected behavior**
No crash and a sliced file ready for importing to my printer 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
Ender 3 base (all metal extruder and stock hotend)

**Desktop (please complete the following information):**
 - OS: [Windows 11]
 - Version [1.6.3]

alldaywarrior, 2023-06-03:
same here #1260

viruz82, 2023-06-03:
This already exists in 1.6.3-beta.

Targolo, 2023-06-20:
Hi,

Same thing for me on most of the projects when I add a raft. I have a P1P and I am using the BBL 0.2mm standard profile + raft.
I've tried on Orca Slicer 1.6.3 and also on version 1.6.2.4 and 1.6.0 of Bambu Studio.
Attached is a sample project.
[Arms.zip](https://github.com/SoftFever/OrcaSlicer/files/11796729/Arms.zip)

closed 2023-11-20 by github-actions[bot]