OrcaSlicer/OrcaSlicer #1222: Keyboard Shortcut customization

iitsalex, 2023-06-01:
**Is your feature request related to a problem? Please describe.**
Many slicers use `backspace` as the default "delete" keyboard shortcut for deleting models. The default on orca is `delete`. This is fine, but would prefer for it to be configurable so I can put backspace.

**Describe the solution you'd like**
A configuration menu for keyboard shortcuts so one can record whatever shortcuts they want. In my case, `backspace` instead of `delete` for removing a model from a build plate.

**Describe alternatives you've considered**
Getting used to using delete (where's the fun in that?)

I may take a stab at this one in my free time, but haven't touched cpp in years.

closed 2023-11-20 by github-actions[bot]

iitsalex, 2024-06-20:
@SoftFever is the standard procedure for re-opening issues like this to just comment on them or should I open a new ticket?

autoferrit, 2025-01-31:
id also like to see shortcut remapping

reboot81, 2025-08-13:
I also would benefit from custom keyboard shortcuts.

Felix14-v2, 2026-09-24:
Implemented in 2.5.0-dev. You can install the [Nightly build](https://github.com/OrcaSlicer/OrcaSlicer/releases/tag/nightly-builds) to test this feature.