OrcaSlicer/OrcaSlicer #1237: Make settings easier to use

julie777, 2023-06-01:
**Is your feature request related to a problem? Please describe.**

I was comparing some profiles and had both Bambu Studio and Orca Slicer open. As I was scrolling through each tab of the process settings I realize how many new settings had been added to Orca Slicer. It was a surprise.

Even before, I had often thought that many of the settings are off screen and require scrolling to see. I realized that many people probably do not realize that some of the settings are available. It was not until I went looking for all places that G code appeared in the actual profile files and saw the post processing scripts that I looked for it in the UI.

This is only going to get worse. It does help when settings are grouped. I noticed at least one BS setting that Orca moved to a more suitable location.

In addition to the number of settings, the button for "Advanced" is often missed by users and I have had to tell many users that they had to select advance to be able to see the setting they were looking for.
 
**Describe the solution you'd like**

I think there are several things that can be done to improve the situation and make working with settings easier.

### Advanced

Advanced is part of a very "full" line. It is on the process line, followed by the line where the process profile is selected. Then there are the tabs, with individual setting below. I think this is part of the reason that users miss it. It does not have any affect on what profiles are available in the chooser. It does change what is shown under each tab. It even hides the speed tab. Somehow it should be in the tab area. I have not thought about it enough to have a location that I prefer.

### Grouping

Tabs are a good idea to group settings into one that are related. Sections with names take that concept even further which is good.

In order to make the vast number of settings more manageable and discoverable, I would like to see the section groups become collapsible. This is a common paradigm that users will recognize and it is already used for the object list. This would "pull" the hidden sections into view when sections are collapsed.

### Favorites

When advanced is off the number of settings shown is much smaller. Ideally this shows new users the settings they might need and the ones that are most easily understood. I suspect many of us have advanced on all the time. It is not likely that we need all the settings all the time, but rather there are settings that we change frequently that are not in the non-advanced settings. 

When the object list is shown it adds a new tab of settings named "Frequent". This is a great idea. However, it is only show for objects. I assume it that is because there is much less room under the objects list to show settings that when you finally find and change one it is added to frequent to make it easier to get to.

Note: settings under the object list would benefit from section collapse/expand because there is limited space below the object list.

I would like to add a favorites tab when global is selected to provide quick and easy access to the settings that I change most often for individual projects.

The reason that I used the heading of favorites was so that we could consider whether a most frequently used list (with a limit set in preferences by the user) or an explicit list chosen by the user explicitly would be better. Regardless of how the list gets created, it seems like an idea that is too good to only use when individual object settings are being changed.
 
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

hliebscher, 2023-06-01:
If long texts are already being copied here, please make sure not to do it twice, and please delete it. 
#1236

julie777, 2023-06-05:
duplicate

closed 2023-06-05 by julie777