OrcaSlicer/OrcaSlicer #1239: Make the sidebar resizable

julie777, 2023-06-01:
**Is your feature request related to a problem? Please describe.**

There are times when it would help if the sidebar was resizable.
For example, under Output/G-code output there is a place to specify the filename format. It is not very usable. (It should really have the text box under the field label)
[image]

At other times I might want to shrink or even hide the sidebar. Possibly when I am working with a large project consisting of many plates. (Until plates can be moved to the plate list in prepare like they are in preview I have them all showing at the same time.) I have a project with 20 plates. Having a larger workspace to manipulate the objects on the plates would be helpful and I could make the sidebar smaller to allow that. 

**Describe the solution you'd like**

1. Provide the ability to change the width of the sidebar. This allows the user to see entire field values and spread out the tabs.

2. Provide a button to hide/show the sidebar so that the workspace can be maximized when working with a large number of plates, or when simulating G-code.

3. Allow the user to choose which side of the window the sidebar is docked to.
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

**Additional context**
Add any other context or screenshots about the feature request here.

Citruzzzor, 2023-11-07:
I agree, this is especially annoying on large 32" and 43" inch screens with 4K resolution.
The inability to make the sidebar wider discourages any desire to switch from Prusa to this slicer
 [photo_2023-10-30_08-43-49](https://github.com/SoftFever/OrcaSlicer/assets/80072365/a26ef5c4-db1e-4aa4-98c2-74ce63ed008d)

tlhintoq, 2023-11-18:
Also on a wide screen and thought the same thing about the sidebar less than 3 minutes into my first launch of the program

foreachthing, 2023-11-18:
Since a bot closes issues quickly, I cast my vote here, too.

tlhintoq, 2023-11-20:
Anti-bot auto-close nudge remark

discip, 2023-12-08:
@julie777
I'm not sure, but maybe this was addressed by #2972?
If so, please consider closing this issue. 😊

igiannakas, 2023-12-26:
This has been implemented

closed 2024-04-02 by github-actions[bot]