OrcaSlicer/OrcaSlicer #1241: Tree Supports are really hard to remove?

dgpmaluco, 2023-06-01:
It seems that tree supports are really thick, and hard to remove? For comparison I use Cura and there the supports almost stay on the bed after removing the print.

Tried adjusting the support settings identically to Cura but that didn’t help. You see in the preview already that there is way to much material between the support and the print

Jonisos, 2023-06-01:
Did you test tree supports and adjusting the "Top interface layers" or the "top interface spacing"?

There are many Youtube videos and Reddit posts on the topic

dgpmaluco, 2023-06-01:
These are my settings, what would you recommend?
https://imgur.com/a/I7uJi5Z

dgpmaluco, 2023-06-01:
Will take a better look at comparing the  settings thanks

closed 2023-11-20 by github-actions[bot]