OrcaSlicer/OrcaSlicer #1247: Generating Gcode Preview

leezoneline, 2023-06-02:
**Is your feature request related to a problem? Please describe.**
In the latest version of Prusa Slicer, there's a feature called G-code thumbnails which can generate costum size of thumbnails of the G-code and preview that on File Explorer or some printers.
**Describe the solution you'd like**
There's an python script on https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview can make old versions of Prusa Slicer to generate thumbnails. I hope that can work but it cannot use in Orca Slicer.
**Describe alternatives you've considered**
This feature can define the size and fomarts of thumbnails.
**Additional context**
[image]
This is the similar result that i want to have in Orca Slicer.

closed 2023-11-22 by github-actions[bot]

GiuseppeSiragusa, 2024-09-15:
FOR ELEGOO GIGA SIDE?

leezoneline, 2024-09-15:
> FOR ELEGOO GIGA SIDE?
> 

For creality ender3