OrcaSlicer/OrcaSlicer #1250: Orca Slicer spontaneously turns off dark mode

julie777, 2023-06-02:
**Describe the bug**

I have had this happen a couple times now. I have no idea what is causing it. I just walked back from the printer and found it looking strange and terrible. I checked preferences and dark mode was off.

I will try to find a way to reproduce it.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.6.3

julie777, 2023-06-02:
I was in the process of creating an issue describing how terrible the UI looks when dark mode is off and went back and turned it off to do a screen capture. It looked fine.

So, even though somehow the preference had automatically switched showing that dark mode was off, the display was not back in light mode. Instead it was in what looked like a color inverted dark mode. This made it mostly black and white and hard to look at. For example, in the settings panel, all the value fields had a black rectangle for the background.

netweaver1970, 2023-06-02:
Happened to me as well. To me, it was (most likely) after a full hibernate + subsequent come-back of the laptop (Win10, Intel embedded video chip + extra nvidia graphics chip)

Splitpulse, 2023-06-07:
I can also say this happened to me. It seems to be when theirs a refresh from sleep mode, another installation, or randomly being on other apps.

beamonster, 2023-06-12:
Same here. Windows 10. Orca Sliver 1.6.3. I noticed this bug long ago on Bambu Slicer. Seems to mostly occur after waking up from sleep or hibernation.

julie777, 2023-06-19:
I have sleep disabled. I also have hibernate turned off when plugged in, which is almost all the time. I cannot recall what the conditions were when it last happened to me.

jwyse, 2023-11-10:
Bump to remove the stale tag; I'd really love to see this issue addressed.  Thanks.

jwyse, 2024-02-12:
Bump to remove stale tag.

closed 2024-05-21 by github-actions[bot]