OrcaSlicer/OrcaSlicer #1252: build on windows using too many resources

julie777, 2023-06-02:
**Describe the bug**

While the build is running, probably building all the deps, I cannot even type without a huge lag because the build is using so much CPU.

At the point when I first looked at processes Git for Windows alone was using 20%. I am not sure why git was running during the build.

There are more than 20 compiler driver processes running at once. Again, I do not know how cmake is configured, but it seems like it should either limit the processes being used or run them at a below normal priority to keep the computer usable.

BTW, I have a capable computer with a core I9 6 cores 12 threads and 32 Gb of memory. That is not the problem.

**Expected behavior**

Build should not stop foreground processes from responding.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version [e.g. 22]

hliebscher, 2023-06-02:
@julie777 
Dear Julie, enough with your comments... I don't care if SoftFever wants to ban me, but you're practically spamming this channel every day with no real issues. I'm sure your input is valuable, but the way you go about it bothers me. I enjoy helping out! Many people here can confirm that, but what you're doing is simply taking away my enthusiasm to continue working on this project. Sure, you joined late, but you're questioning everything we've worked on with joy. Nothing seems to be okay for you, and you want things exactly as you imagine them. If you have a suitable development team, then let them handle it. But reading your lengthy texts in the issues every day is too exhausting for me. I'm sorry for being so blunt here. It has been a pleasure for me until now to contribute my time to one of the coolest projects and share it with the rest of the developers.

julie777, 2023-06-02:
@hliebscher 
I do appreciate your responses. Often you enlighten me. I do not mind you being blunt and sharing how you feel. I am not trying to have a negative affect on you or the project.

However, I think that reporting real issues or requesting features is not a bad thing. The project owner can choose how to prioritize the issue backlog.

I cannot see how reporting how the build is affecting my computer is questioning anything that you have done.

In addition, you do not have to read my issues. You are welcome to ignore them. I will admit some of what I write is lengthy. I have tried to use discussions for that. In a bug report, what you call lengthy I might call thorough. In all my years working as a software engineer, I have never once felt like a bug had too much information. The opposite was probably true more than 80% of the time.

How can I not irritate you while continuing to contribute?

closed 2023-11-22 by github-actions[bot]