OrcaSlicer/OrcaSlicer #1253: Klipper: Unknown command G3 & G17

djeZo888, 2023-06-02:
I have used OrcaSlicer to slice some print model for custom Klipper modded printer. The print came out just fine, but I was getting bunch of:

`
...
Unknown command:"G3"
Unknown command:"G17"
...
`

Is this a problem, a setting I should enable/disable when slicing?
Thanks

hliebscher, 2023-06-02:
Have you the latest version and enable Arc support?

djeZo888, 2023-06-03:
Arc fitting is disabled - that is default Klipper profile in OrcaSlicer.

djeZo888, 2023-06-03:
Oh, I just found this comment on reddit:

`Regarding the G3/G17 gcode, there are two solutions: 1. Enable gcode_arcs in klipper config. 2. Change Z Hop type to normal (spiral uses G3 I believe).`

And one of the big reasons for using OrcaSlicer is Spiral Z Hop - I have it enabled. So I have to make sure that my klipper supports these commands.

tehbiga, 2023-07-31:
I'm having this same issue after just switching to Orca Slicer and realized even with Z Hop set to `Normal` it uses arcs when I have arc fitting disabled. I considered just enabling arc support in Klipper, but then realized it's just Orca still doing arc based Z Hop when set to `Normal`.

I don't want Z Hop anyway so I just set the distance to zero and it stopped generating g-code for hops.

closed 2023-11-22 by github-actions[bot]

danshoop, 2023-11-28:
Issue persists. Should be reopened.

guilhermepozo, 2023-12-10:
same here, G3 command everytime

oernster, 2024-01-28:
Looked up klipper gcodes: here you are...

[gcode_arcs]
The following standard G-Code commands are available if a gcode_arcs config section is enabled:

Arc Move Clockwise (G2), Arc Move Counter-clockwise (G3): G2|G3 [X<pos>] [Y<pos>] [Z<pos>] [E<pos>] [F<speed>] I<value> J<value>|I<value> K<value>|J<value> K<value>
Arc Plane Select: G17 (XY plane), G18 (XZ plane), G19 (YZ plane)

gfBarreto, 2024-09-27:
Provavelmente já encontraram a solução, mas resolvi informar que estava tendo o mesmo erro aqui...
Lembrei que foi depois de ajustar no orca, em retração, deixei em auto e provavelmente estava fazendo movimentos arc.

ChD0903, 2026-10-13:
Lösungen

• Z-Hop umstellen: Stelle im Slicer (z. B. OrcaSlicer unter Filament-Einstellungen -> Setting Overrides oder bei den Extruder-/Z-Hop-Optionen) den Z-Hop-Typ von Spiral oder Auto auf Normal um und deaktiviere spezielle Bogen-Retractions.
• Klipper anpassen: Aktiviere in deiner printer.cfg die Unterstützung für Bögen, indem du [gcode_arcs] hinzufügst (falls noch nicht geschehen).
• Ignorieren oder auskommentieren: Da Klipper den Befehl oft einfach ignoriert, kannst du die Meldung bei reibungslos laufenden Drucken ignorieren.