OrcaSlicer/OrcaSlicer #1257: Nerfed Arc Generation: Changing VFR to the wrong value kills arc generation

psiberfunk, 2023-06-03:
This issue is similar to #1256 , but different because it is even WEIRDER.  It might be a related cause, however.  It sounds insane, but I have a case here where merely changing the volumetric flow rate for a filament, and no other variables, stops arc generation for the outer perimeter (inner perimeters not working , per #1256 ).

**Describe the bug**
Under the conditions in the attached project files the slicer is NOT generating outer perimter OR inner perimeter arcs when the VFR for a filament is changed from 21-->12 mm^3/sec.    Hard to believe, but true! Bambuslicer 1.6.2.4 does not have this problem (or #1256).  Problem exists at least as far back as V1.4.5 in Orca. Here's the proof:

**3mf File for This Bug**
See attached ZIP file with:
[VFRIssue.zip](https://github.com/SoftFever/OrcaSlicer/files/11641687/VFRIssue.zip)

- _Chillmaster_VFR21_ArcsOnOUTERPerimetersOnly_OrcaProject.3mf_   (VFR=21)
- _Chillmaster_VFR12_NoArcsOnPerimeters_OrcaProject.3MF_ (Same settings but VFR=12)
- Screenshots below also in ZIP
- Source STEP file, so that you may see that this is not a file import resolution issue.

**To Reproduce**

- Open the attached project files in OrcaSlicer 1.6.3 and slice, observe, as shown in the screenshots below, that G2/G3 arcs are generated on outer perimeters only when VFR=21, but NOT when VFR=12.  Bizzare!
- Slice the same project file in Bambu Slicer.  Observe, as shown in the screenshot, that G2/G3 arcs are generated for BOTH Inner+outer perimeters, even when VFR=12.

**Expected behavior**
Orca should slice this file like Bambu does, providing arcs for inner+outer perimeters for the circular features in the file.

**Screenshots**
Screenshots here show Orca for VFR=21, VFR=12, and Bambu for VFR=12.

Orca VFR=21 [image]
Orca VFR=12 [image]
Bambu VFR=12 [image]

**Printer model**
- Bambu X1  (but i'm not sure this is printer-specific)

**Desktop (please complete the following information):**
 - OS: Windows
 - Multiple versions, problem exists in v1.6.3,  v1.5.0, v1.4.5  (only tested these 3 before I gave up)

SoftFever, 2023-06-08:
The way how PrusaSlicer 2.6's overhang slowdown approach is not playing nicely with arc fitting.
I'm going to rewrite this part. For now, as we workaround you can use the classic mode if you prefer to have more arc moves.
[image]

psiberfunk, 2023-06-08:
IIRC bambu also re-wrote this themselves already in their slicer, did they not ?

cpasoft, 2023-06-08:
Not only is the arc generation broken, but the overhang slowdown is also doing strange things...
I think this issue is also related to #1217

psiberfunk, 2023-07-24:
This one still really hurts and drastically hurts print quality on bambu printers at higher speeds because their motion controller doesn't do nearly as well with lots of little lines vs arcs.  Any chance for a resolution ?

psiberfunk, 2023-07-28:
Update based on nightlies testing: This would appear to be fixed as of https://github.com/SoftFever/OrcaSlicer/actions/runs/5693025113 at least ?

psiberfunk, 2023-08-26:
@SoftFever  this is now broken again as of 1.6.6 ... it seems the re-write did not account for this problem ?

PR #2942 (2023-12-01): Improve arc fitting when new overhang slowdown is used
Now arc fitting should work properly for internal walls.
Also max volumetric speed won't affect arc fitting results anymore.
[image]

Fixed #1257 #1256 #1329 #1257 #1257