OrcaSlicer/OrcaSlicer #1260: Critical error, restart program over and over again

alldaywarrior, 2023-06-03:
Slicing object with .12 and .08mm layer height and 3 raft layers gives critical error and asks me to restart program.
Doing what the error tells me to do does not work and it sometimes give me different errors. Never had this before. The file that I try to print is made by myself and is safe.

**To Reproduce**
Setting layerheight to .08 or .12mm and raft 3 layers.

**Screenshots**
[image]

**Printer model**
Bambulabs p1p
**Desktop (please complete the following information):**
 - Windows 10
 - Orcaslicer 1.6.3

hliebscher, 2023-06-03:
do you have a saved 3mf that you can send as zip?

HunterG6700, 2023-07-26:
I have the same issue when slicing with a raft.

closed 2023-11-22 by github-actions[bot]