OrcaSlicer/OrcaSlicer #1261: Bug:  Incorrect G-Code Generation in Version 1.6.3

philippwaller, 2023-06-03:
**Issue Description:**
In the current version 1.6.3 of the software, I have observed a bug related to G-Code generation. The software generates G-Code instructions that command the printer to move along paths that are not defined in the model. This issue occurs specifically in the OrcaSlicer, while the BambuSlicer does not exhibit the same problem.

**Steps to Reproduce:**
Install and launch version 1.6.3 of the software.
Load the provided STL file.
Slice the model using the OrcaSlicer.
Review the generated G-Code instructions.

**Expected Behavior:**
The generated G-Code should accurately represent the model geometry, with no extraneous or unexpected movement commands.

Actual Behavior:
The generated G-Code contains instructions for the printer to traverse paths that are not present in the model.

**Attachments:**
OrcaSlicer
[image]

BambuSlicer
[image]

[Sample File.stl](https://file.io/zcju9jO1hCAm)

RaySajuuk, 2023-06-03:
I sliced your file in OrcaSlicer Nightly ([c675d97](https://github.com/SoftFever/OrcaSlicer/commit/c675d979a746390a3243eadf97781fb34525aa27)) did not see any issues.

[image]

philippwaller, 2023-06-04:
I just tried the latest snapshot and I get the same result. The problem will be resolved with the next release. Thank you very much.

closed 2023-06-04 by philippwaller

RaySajuuk, 2023-06-04:
Go to the action tab to download the latest nightly build or compile it independently. Or you can compile it yourself.