OrcaSlicer/OrcaSlicer #1268: Set model name when sending to Bambu X1

Zodiase, 2023-06-04:
Orca Slicer 1.6.3.

## Problem

For example, I set a custom project name ("Snapfinity 5x5 HT 0.4 PLA") when sending to printer:
[image]

When I go to the printer, in the Files grid view I can see my custom project name "Snapfinity 5x5 HT 0.4 PLA". But because the grid view can only show so long of a name, I tap the file to open it to see its name.

This is where I see "1x4" as its name. I couldn't for my life understand where that came from as I don't recall naming anything that way. Finally I recall perhaps the first model I dragged into the project had name "1x4.step". And I was able to find the "Project" tab (that I never used before) where it clearly says "Model name: 1x4".
[image]

## Expectation

I'm not sure what the use of the "Project" is as it doesn't appear anything is editable. Also it doesn't quite make sense to use that "Model name" when I send any of my 4 plates to the printer.

I'd like the projects sent to the printer to use the custom name I set in the "Send to Printer SD card" interface. This is not just the name displayed in the grid view but also the name displayed when I tap and open it.

Zodiase, 2023-06-04:
Oddly, I tried to write a reproduction but I couldn't get the "Model name" in the Project tab to change to anything non-empty, no matter what I have tried so far.

closed 2023-11-22 by github-actions[bot]

CoReYeDe, 2024-02-01:
Would like to reopen, because i have the same behavior..
STL exported from Fusion360, imported to orcaslicer..
Dont know, where this Information is get from and how to change it.
Also "Bambu Weihnachtshütte" - What is this?

[image]
[image]
[image]