OrcaSlicer/OrcaSlicer #1269: Profile Inheritance

dguisti, 2023-06-04:
**Is your feature request related to a problem? Please describe.**
No specific "problem," per-say, but when calibrating the printer there are certain values (like flow rate) that are less likely to change than other filaments. For example, if I have a "Generic PLA" filament, a "[Brand] Generic PLA" filament, and then a "[Brand] PLA Blue" filament profile and want to change the temperature or flow ratio for all of them, I have to go through and manually change all three profiles (and for every other roll of PLA I have).

**Describe the solution you'd like**
I would like profiles to be able to inherit from each other. I can see that the filament JSONs have an "inherits" parameter. In PrusaSlicer, if you modify this parameter and then change values in one filament, it will allow you to click an arrow to revert the parameter to the parent value. I'm not sure why (or if it was intentional for) this feature to be disabled in OrcaSlicer, but it doesn't work that way—if you change a parent filament, all filaments that inherit from it don't change, although when you re-save child filaments any non-conflicting information _is_ removed from the storage file.

But not only would I like this to do similar inheritance like PrusaSlicer does, I would like two additional features to make profile management easier:
1. *Automatic updating of sub profiles* — When a parent profile is changed, all sub-profiles should automatically be updated to match. Child profiles should have an option to revert either individual settings or the entire profile to the values from the parent profile.
2. *A profile management interface* — I would enjoy having a dedicated menu to rearrange profiles and visualize their inheritance from each other. This would allow you to easily move filaments between different groups and parents to make managing large varieties of settings easier.

**Additional context**
I'd like to note that I would be happy to (and am planning on, unless someone else does first) implementing these features myself; however, I wanted to get some context from people who are more up-to-date and familiar with the project on what changes have already been made in this direction and what further modifications would need to be made to allow for it.

dguisti, 2023-11-15:
Status update: Bambu Studio has implemented custom filament profiles in their [1.8 beta](https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57). After this feature gets ported over to Orca that would probably be a good time to revisit this and see if we can add options for inheritance. Since this issue hasn't garnered substantial public attention, I might take that modification into my own hands after we see what the roadmap looks like for porting new Bambu Studio features.

closed 2024-02-23 by github-actions[bot]

Ninjadude101, 2024-10-30:
Did anything ever come of this? I would like to set up inheritance in my profiles too so that I don't have to make the same change to a load of profiles whenever I tweak a setting, and find it wild that this can't already be done as far as I can tell.