OrcaSlicer/OrcaSlicer #1271: Volumetric Flow Rate speed Derivation

Lt-Kammello, 2023-06-05:
SUGGESTION: setup volumetric flow rate printing like in SuperSlicer

In SuperSlicer you can set default_speed to "0" allowing the filament_max_volumetric_speed to control default_speed and other derived speed.
This is very useful in many cases, mainly if you have one print Process wich is used with an array of filaments having same printing setting except flow rate.
Exampl: i have a printing process (profile9 wich is common both for PA-CF and some harder Flexible filaments beracuse thay share many setttings like for example seam_position = nearest.
Since volumetric flow rate do not control deafult_speed except acting as a limiter, i have to duplicate many printing processes using different speed sets.
Being able to use filament_max_volumetric_speed as a global volumetric speed setting instead of a limiter would allow to reduce drastically the number of duplicated printing processes.

filament_max_volumetric_speed is also useful in optimizing printing speed without impact on quality.

Hunkoys, 2023-06-08:
Isn't this already the case. I actually use Max Volumetric Rate as a speed limiter. My print profile would have the max the printer can handle.

charredchar, 2023-07-23:
This is one of the main things keeping me from moving away from SuperSlicer. You can set your Maximum Volumetric Flow Rate then when `Default Speed` is set to `0` it will auto calculate the speed based off the flow rate. Most other speeds can then be set to a percentage based on the `Default Speed`, ie outer walls being 75% or infill being 125%, and it will handle everything on its own without the need to tweak every single speed every single time you want to do something different.

charredchar, 2023-11-06:
Still hoping this could happen...

closed 2024-02-19 by github-actions[bot]