OrcaSlicer/OrcaSlicer #1275: Can we add a universal search feature like in the bar on PrusaSlicer? Coming from other slicers it can be hard to remember what settings are where.

dvomang, 2023-06-05:
**Is your feature request related to a problem? Please describe.**
I would like to be able to search within any setting when searching in Orca. If I forgot where pressure advance is, or line width etc I could search for them globally.

**Describe the solution you'd like**
[image]

liftbag, 2023-06-07:
> **Is your feature request related to a problem? Please describe.** I would like to be able to search within any setting when searching in Orca. If I forgot where pressure advance is, or line width etc I could search for them globally.

The search feature is already present, but not universal. You can search in the process presets, but not in the filament or printer settings. In fact it could be a useful improvement for those switching to Orca Slicer for the first time.

[image]

dvomang, 2023-06-07:
Not related to a bug.

I just think it would make more sense if all of the settings were searchable rather than just the processes. There are so many settings and sometimes it’s easier to just search for what you are looking for rather than hunt through every setting looking for it.

julie777, 2023-06-09:
I have to agree with @dvomang, being able to search all settings at the same time using `<ctrl>F` like Prusa Slicer is so much better. I hate having to search in 3 places and two of them require opening a separate dialog.

BTW, I submitted this request to Bambu Studio quite some time ago.

jeronbernal, 2023-07-15:
this would be a great feature! i agree, it would really help with switching slicers

YanceyA, 2023-07-16:
I too would like this feature.

bvrielink, 2023-07-19:
Coming from Prusa Slicer I really miss the CTRL-F function. Hope it will be added.

closed 2023-11-22 by github-actions[bot]

alfondehesa, 2024-01-31:
Same here. Would love the CTRL-F function to search all settings (for processs, filament and printer all together).

swdwi2, 2025-05-16:
Same here.  Miss this feature switching from superslicer