OrcaSlicer/OrcaSlicer #1277: Assign different printers  and settings to different build plates within a single project

dyfiastro, 2023-06-06:
**Is your feature request related to a problem? Please describe.**
I have multiple printers with different build volumes and settings that I want to use within a single project

**Describe the solution you'd like**
The ability to add build plates but assign each build plate a printer and profile

**Describe alternatives you've considered**
loading multiple instances of the program, one per printer but does not work when trying to arrange parts across build plates

hliebscher, 2023-06-06:
I have running multiple instances so It works perfectly
If you add all parts to any instances, so it's easy to order the parts to the plates

dyfiastro, 2023-06-07:
> I have running multiple instances so It works perfectly If you add all parts to any instances, so it's easy to order the parts to the plates

This does not seem to be working for me at least the way I want it to.
Let me explain a little better.

I put X Parts on Plate A. I assign Printer 1,Filament 1 and Setting 1 to those.
I create a new plate B with Parts Y. I assign Printer 2, filament 2 and settings 2 to those.

As soon as I switch back to Plate A all the settings I set to Plate A are now gone and its assigned all the stuff I have just assigned to Plate B.

hliebscher, 2023-06-10:
the pinter and settings are assigned to all plates ... if you need 2 printer and switch between you have to open 2 instances

julie777, 2023-06-19:
> I have running multiple instances so It works perfectly If you add all parts to any instances, so it's easy to order the parts to the plates

I do not see how running multiple instances solves the problem. 
I have a project with multiple plates.
I want to assign profiles nozzle .2, PLA, .16 quality to one plate.
I want to assign profiles nozzle .4, PETG, .4 strength to another plate.
I want to save the project with these settings so that when I open the project later it remembers the settings for each plate.

This also allows me to update one object in CAD and replace it on a plate and have that object be printed with the settings for that plate. Currently I have to set the settings for the object individually as modifications to the profiles set for the project. Then when replacing the object with a newer version I have to redo all the object settings.

jagerman, 2023-07-31:
I would love this feature.  I often have a few parts to print and will split them between a Trident 300mm-bed printer, and a vastly different sized V0 120mm-bed printer.  I'd love it if I could add a "Trident plate" associated with printer 1, and a much smaller "V0 plate" associated with printer 2, in the same project so that I could arrange parts across the two plates to decide how I want to split up the print job across the two machines, and then have the Print button on each plate send the job to the appropriate printer for that plate.

(Currently even just switching to a printer with a different bed size when you have things on plates just gives you a mess of parts positioned badly.)

closed 2023-11-22 by github-actions[bot]

jagerman, 2023-11-22:
Bad bot.

WinstonLeWolf, 2024-03-17:
Please re-open.  This would be a great feature to have.

unabletoconnect2429, 2024-05-30:
second this. Switching between a bambu p1 and a voron 2.4 all the time. Would be great to assign plates per printer

AnisBoubaker, 2024-06-30:
Please re-open, this would be a great feature to print a project on multiple printers at once.

NiHuShu, 2024-07-21:
Bump, I have projects where some of the parts are printed on P1P and others on A1 mini, it'd be great if I could keep all the parts in a single project.

5hiftyy, 2024-07-23:
Adding my support. Even something as simple as a rugged box requires two colours of ABS, and a TPU seal. Having three separate materials within the same project, organized by plate, would be an amazing improvement to the workflow and productivity.

YairHH, 2024-09-05:
I also want to express that this would be an incredibly useful feature. being able to assign to each plate their own "Printer,"  "Filament," and "Process" selection. I often work with multiple printers at once; I do currently open an instance per printer.

rickbeec, 2024-11-05:
I was actually kinda' shocked that this wasn't already a feature. Perhaps it's a heavy lift on the development side of things? I can tell you from a production perspective, where I am printing a larger, multi-component piece, it would be awesome to be able to assign particular plates to different machines. It keeps the project contained, but allows each printer to be used for it's specific strengths.

torchddv, 2024-11-17:
I came here looking for the answer to this question as I also assumed it was a feature that I couldn't find. After all, if one adds additional filaments there's a column labelled "filament" beside the object names. So please put me down as another one who would like to see this added. 

As a work around to change filaments, one can do a colour change on the first layer of the print of one specific plate -- but one cannot change the process. ie: only the filament properties (eg: cooling) are affected but not process properties (eg: speed).

aguthard, 2025-01-11:
I would like to throw my hat into the ring and also push for this feature request to be implemented.
These days, more and more people are working on a large project with multiple printers at the same time, so this solves a fundamental problem.