OrcaSlicer/OrcaSlicer #1281: Gcode template broken (for Biqu B1 at least)

Duncro, 2023-06-06:
**Describe the bug**
When slicing I I get the error :
```
Failed to generate G-code for invalid custom G-code.
machine_start_gcode Parsing error at line 3: Not a variable name
M140 S{material_bed_temperature_layer_0}; Set Heat Bed temperature
       ^
Please check the custom G-code or use the default custom G-code.
```

**3mf File for This Bug**
irrelevant

**To Reproduce**
Setup printer for Biqu B1, Biqu B1 generic PLA, try slice any model

**Expected behavior**
It should slice successfully

**Printer model**
Biqu B1

**Desktop (please complete the following information):**
Windows 10
Orca slicer 1.6.3

Duncro, 2023-06-06:
[BIQU B1 (0.4 nozzle).json.txt](https://github.com/SoftFever/OrcaSlicer/files/11667277/BIQU.B1.0.4.nozzle.json.txt)
This fixes it

cron410, 2023-06-28:
For completeness, on Windows you would replace the relevant file here:
C:\Users\___________\AppData\Roaming\OrcaSlicer\user\default\machine

closed 2023-11-22 by github-actions[bot]