OrcaSlicer/OrcaSlicer #1282: Inserting PAUSE Gcode before PRINT_END Gcode

PhilBaz, 2023-06-06:
**Describe the bug**
A clear and concise description of what the bug is.

Its inserting PAUSE Gcode before PRINT_END Gcode

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

slice

**Expected behavior**
A clear and concise description of what you expected to happen.

No pause Gcode

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Printer model**

Voron 2.4

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

Windows 10

PhilBaz, 2023-06-06:
Ahhh, this is the filament end Gcode. But I'm only using one filament ?

hliebscher, 2023-06-06:
You can add a if command like if tool changes >0 then ....
Ignore pause with 1 filament 
Sample in snapmaker j1 filament gcode

PhilBaz, 2023-06-07:
@hliebscher 

Can you paste the code here? I looked but couldn't find it. It goes right in the filament end Gcode box? Didn't know we could put conditionals in there.

Thank you!

hliebscher, 2023-06-10:
@PhilBaz 

{if total_toolchanges < 1};  one extruder /Filament
;nothing
{else}; more extruders
PAUSE_BEEP
{endif}

PhilBaz, 2023-06-10:
@hliebscher 

Thank you!  

I don't know if I should close this issue or leave it open to be addressed as a setting or in the documentation?

closed 2023-11-22 by github-actions[bot]