OrcaSlicer/OrcaSlicer #1286: No UI shows after loading screen disappared

XianmingFeng, 2023-06-07:
**Describe the bug**

v1.5.0 works fine but both v1.6.2 and v1.6.3 cannot open the main screen.

v1.6.2 and v1.6.3 show the loading profiles screen and then disappare. Then nothing comes up, no main screen. I can see no OrcaSlicer process in task manager.

After the failed lauching of v1.6.2 or v1.6.3, profiles saved in v1.5.0 disappared. No costom profile can be found when I open v1.5.0 again.

**To Reproduce**

double click orcaslicer.exe icon or right click the icon then click run as admin.

**Expected behavior**

I expect there should be a main screen to work on my projects. And the profiles should not disappare.

**Desktop (please complete the following information):**
 - OS: Windows 11 22H2 build:22621.1702
 - Version 1.6.2 1.6.3

psiberfunk, 2023-06-07:
You don’t mention: did you try installing the required dependencies in the readme?

XianmingFeng, 2023-06-10:
The dependencies have been installed. I checked by try to install them again and yes they reported me "already installed". Does the version of the dependencies matter?

XianmingFeng, 2023-06-11:
> What happens if you try launching from a command line .. what errors do you get ?

No error, actually nothing returned from command line.

closed 2023-11-22 by github-actions[bot]