OrcaSlicer/OrcaSlicer #1292: Tool change system/multi extruder not supported

vito30x, 2023-06-08:
I use a Custom made printer with a Tool change system like the BTT Hermit Crab.

Rob65, 2023-06-08:
I would also love this feature. If being implemented, please also consider to add overrides for speed/acceleration/jerk. Especially for tool changer with different (weight) tools, this would be very nice.

You might be able set up multiple filament/extruder combos and add the code for this in the advanced tab by adding custom G-code. That is not the most user friendly way but this may work.

hliebscher, 2023-06-10:
this is the bambustudio problem ... they have only 1 Toolhead/extruder in software... so if you use multicolor only one nozzle size is supported

Rob65, 2023-06-10:
This is a very bad comment. If the fact that this is not supported in bambustudio means this will not be implemented in OrcaSlicer then I see limited use for OrcaSlicer.

vito30x, 2023-06-10:
> this is the bambustudio problem ... they have only 1 Toolhead/extruder in software... so if you use multicolor only one nozzle size is supported

I was hoping that this will implemented, because it´s a hybrid also from Prusa slicer and there you can setup multi extruder and nozzle sizes.

hliebscher, 2023-06-11:
I hope softfever make this possible ... 
i need this for my IDEX Printers

EiNSTeiN-, 2023-06-12:
This would be useful :+1:

danik-ik, 2023-10-10:
This seems to be exactly the feature that will turn me away from the Orca slicer at the selection stage. However, I will keep an eye on the situation as it develops.

closed 2024-01-17 by github-actions[bot]