OrcaSlicer/OrcaSlicer #1293: 'A G-code path goes beyond plate boundaries' after exporting and importing sliced 3mf for Prusa

Darthmonkey, 2023-06-08:
**Describe the bug**
A clear and concise description of what the bug is.
After adding items to the build plate, then 'Export plate sliced file' as 3mf, then importing them into a new Orcaslicer instance, an error shows stating 'A G-code path goes beyond plate boundaries' when using the Prusa profile.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior
1. Change the Printer to 'Prusa on Octoprint'
2. Add 3mf files to a build plate
3. Ctrl-G to export the plate sliced files as 3mf
4. Close Orcaslicer
5. Double click the 3mf file to open in Orcaslicer.
6. Receve the error 'A G-code path goes beyond plate boundaries'

**Expected behavior**
A clear and concise description of what you expected to happen.
Steps 1-5 above, but without the error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
Prusa Mk3s on OctoPrint

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]
Windows 11

SoftFever, 2023-06-08:
@Darthmonkey 
Is it convenient to share the project file as well as the sliced file?

Darthmonkey, 2023-06-09:
Sorry, I'm not completely sure what you mean by sliced file or project files. I have tried to interpret the best I could.
https://1drv.ms/f/s!AkvX-bEguFc_jNoekuZ5kufP2BNk4Q?e=rkHXdH

closed 2023-11-22 by github-actions[bot]