OrcaSlicer/OrcaSlicer #1298: [Question] KAMP Bed mesh not using Exclude Object

Semorphim, 2023-06-08:
I'm in the Process of switching from SuperSlicer to Orca.
After transfering my Settings for Filament, RatRig V-Core3 and Printsettings im trying to print a file.
At first everything starts as normal but the adaptive Bed mesh is not generated properly. The mesh is created over the whole buildplate and not localized to the file.

0 object points, clamping to bed mesh [(20.0, 20.0) (465.0, 460.0)] is said in the Console.
I've tried the "Fix" from kyleisah's KAMP FAQ whith using M117 to trick Klipper into seeing the EXCLUDE_OBJECT_DEFINE Statement with no Luck.

Any pointers to where my problem could come from?

hliebscher, 2023-06-08:
Klipper or RRF Printer Firmware?

Semorphim, 2023-06-08:
That was fast, thank you so much!

klipper
v0.11.0-205-g5f0d252b

Machine start G-Code

M117
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]

Start Print G-Code macro

[gcode_macro START_PRINT]
description: Start print procedure, use this in your Slicer.
gcode:
  CLEAR_PAUSE
  SAVE_GCODE_STATE NAME=start_print_state
  # Metric values
  G21
  # Absolute positioning
  G90 
  # Set extruder to absolute mode
  M82
  # Home if needed
  MAYBE_HOME
  M117 Heating bed...
  RESPOND MSG="Heating bed..."
  # Wait for bed to heat up
  M190 S{params.BED_TEMP|default(printer.heater_bed.target, true) }
  # Run the customizable "AFTER_HEATING_BED" macro.
  _START_PRINT_AFTER_HEATING_BED
  # Run the customizable "BED_MESH" macro
  # _START_PRINT_BED_MESH
  BED_MESH_CALIBRATE
  # Start heating extruder
  M104 S{params.EXTRUDER_TEMP|default(printer.extruder.target, true) }
  # Run the customizable "PARK" macro
  _START_PRINT_PARK
  # Wait for extruder to heat up
  M117 Heating Extruder...
  RESPOND MSG="Heating Extruder..."
  M109 S{params.EXTRUDER_TEMP|default(printer.extruder.target, true) }
  # Run the customizable "AFTER_HEATING_EXTRUDER" macro.
  _START_PRINT_AFTER_HEATING_EXTRUDER
  M117 Printing...
  RESPOND MSG="Printing..."
  RESTORE_GCODE_STATE NAME=start_print_state
  # Set extrusion mode based on user configuration
  {% if printer["gcode_macro RatOS"].relative_extrusion|lower == 'true' %}
    M83
  {% else %}
    M82
  {% endif %}
  G92 E0

hliebscher, 2023-06-10:
have you Label Objects activated?
[image]

Semorphim, 2023-06-11:
OrcaSlicer 1.6.3
Label Objects and Exclude Objects is both enabled.

Here is the start of a sliced file:

; EXECUTABLE_BLOCK_START
EXCLUDE_OBJECT_DEFINE NAME=Nero.stl_id_0_copy_0 CENTER=250,250 POLYGON=[[224.917,250.185],[232.061,232.379],[249.788,224.917],[267.595,232.167],[275.082,249.894],[267.648,267.78],[250.079,275.083],[232.22,267.965],[224.917,250.185]]

Semorphim, 2023-06-12:
After disabling Exclude Object in OrcaSlicer Kamp is using the mesh again

closed 2023-06-12 by Semorphim