OrcaSlicer/OrcaSlicer #1302: Issue slicing certain models

Edzi11, 2023-06-08:
### Discussed in https://github.com/SoftFever/OrcaSlicer/discussions/1299

<div type='discussions-op-text'>

<sup>Originally posted by **Edzi11** June  8, 2023</sup>
As stated I'm having some issues slicing certain models when using supports. It freezes after getting to 65%. The only time it doesn't
 is when I'm using slim tree supports or default normal supports. Oddly enough cura has no issue with the models and slices it within minutes. Attached is the model that isnt working correctly. File is about 1.8mb and about 40k triangles but I have sliced bigger files in cura without issue. any suggestions as I want this to be my main slicer but not if it cant slice every file i throw at it like cura does

[image]
</div>

hliebscher, 2023-06-10:
it works with 1.6.3 Mac it takes 3 sec !      but the stl model has errors

hliebscher, 2023-06-10:
[image]

Edzi11, 2023-06-10:
Yes I did notice I had a few manifold errors but on Windows you can repair
it within the slicer. Do me a favor when generating the supports using slim
supports change the diameter of the supports to five and the distance to
two. Let me know if it's still generates for you. So if I don't change any
variables on the supports and just use the default tree support slim it
does generate but it takes a solid 20 minutes to do so. But changing the
settings I just advised doesn't slice for me even after an hour

closed 2023-11-22 by github-actions[bot]