OrcaSlicer/OrcaSlicer #1303: Orca slicer support is solid, printing 100% infill

manpres2, 2023-06-09:
Hi,

I recently started using ORCA slicer and have experience with Cura and prusha slicer. I have tried to find the option if i can reduce the infill for the supports but its always printing solid and even shows solid in the preview. I don't see an option to change this. I want prusha like tree support if there is an option in the slicer.
[image]

atl285, 2023-06-09:
Change the Support **style** to which tree style you want:
[image]

hellfirehd, 2023-06-12:
I am seeing the same thing. 

- Version: v1.6.3
- Type: Tree (auto)
- Style: Tree Slim

[image]
[image]

fellipute, 2023-06-12:
same here, any clues?

atl285, 2023-06-14:
After changing Support **Type** and **Style** in OrcaSlicer 1.6.3 it looks fine for me:
[image]

manpres2, 2023-06-15:
[image]
  its still same

closed 2023-11-22 by github-actions[bot]

lin-ycv, 2024-01-08:
v1.9 still an issue
[image]